realsenseai/librealsense #3931: Unable to install realsense package

pringithub, 2019-05-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Operating System & Version |   {Ubuntu 16.04}  | 
| Platform                                 | RPi 3 B+ |

### Issue Description
Hello,
I am trying to install the dkms linux package on my RPi using this guide: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

I am unable to add the remote server to my list of repositories. Every time I run the add-apt-repository command, it seems like I am waiting for awhile with no print statements, and then it will either exit or spew garbage. Checking /etc/apt/sources/list.d/ confirms no repo has been added. Moreover, running 'sudo apt-get install librealsense-dkms' reveals it can not find the package. Is there an issue with this remote server at the moment?

Thank you in advance

ev-mp, 2019-05-07:
@pringithub hello,
Regardless of the reported issue with the server's certificate,  the Debian distribution packages are available for x86/x64 platforms only. 
This somewhat a recurring request and we're looking into additional ways to streamline the setup for non-Intel platforms, but for now you'll need to build the SDK from sources.

There are several Raspberry-tailored installation guides contributed both by the community and Intel. Follow up the question in #3895 for details.
On our side we'll review the site's documentation to make sure that this statement is embedded.

pringithub, 2019-05-14:
Thank you - I had to do some digging to find that ISA compatibility is an issue.

closed 2019-05-14 by pringithub