realsenseai/librealsense #3935: How to built and run ready-to-hack script?

Ahad19931, 2019-05-08:
| Camera Model                                     D435i
| Firmware Version                                 05.11.06.200 
| Operating System & Version |              Linux (Ubuntu 16.04 LTS) 
| Kernel Version (Linux Only)|                4.15.0-48-generic
| Platform|                                               PC 
| SDK Version                                         2.21.0                          
| Language                                             python/C                          
| Segment			                       Robot                           

Hi,
I want to do obstacle avoidance with the D435i camera for which I need to get the depth information of the pixels. Thus, I need to run the ready-to-hack script but I am not quite sure how to do that. I need assistance to run this script in order to get depth info.
Thanks

RealSenseCustomerSupport, 2019-05-29:
----------------------------------------------

Hi @Ahad1993,

I am not sure about "ready-to-hack" script. Can you elaborate more?

RealSenseCustomerSupport, 2019-06-30:
----------------------------------------------

Hi @Ahad1993,

Any update? If no further support required, will close this.

closed 2019-07-12 by RealSenseCustomerSupport