realsenseai/librealsense #3936: Librealsense Android (librealsense2.so): Which std library is used?

xtrawurst, 2019-05-08:
Which std library is used to compile librealsense2.so? We have an Android app with a large native codebase and need to make sure we have compatibility (must not mix different std libraries as per https://developer.android.com/ndk/guides/cpp-support).

Looking into librealsense2.so with objdump reveals
```
...
NEEDED               liblog.so
NEEDED               libm.so
NEEDED               libdl.so
NEEDED               libc.so
SONAME               librealsense2.so
...
```
So it looks like either no C++ std library was used or static linking was used (more likely).

Also: is there any guide on how to compile librealsense for Android, in case we need to use a different std library?

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

You can add these lines to the end of <librealsense>/wrappers/android/librealsense/CMakeLists.txt
```
message(STATUS "ANDROID_STL: ${ANDROID_STL}")
message(STATUS "ANDROID_TOOLCHAIN: ${ANDROID_TOOLCHAIN}")
```
and build it.

In my side, got these message in the build log.
ANDROID_STL: c++_static
ANDROID_TOOLCHAIN: clang

xtrawurst, 2019-05-09:
@RealSenseCustomerSupport 

Thanks. This is what I thought. I think it would be good if several versions of librealsense2.so were provided, with different STLs.

Also, statically linking the STL on Android is generally not recommended [(see here)](https://developer.android.com/ndk/guides/cpp-support#static_runtimes)
> In general, you can only use a static variant of the C++ runtime if you have one and only one shared library in your application.

The most common STLs in use for Android shared libs are gnustl_shared and libc++_shared, so it would be good if at least two versions of librealsense2.so would be provided, one for each STL.

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

https://developer.android.com/ndk/guides/cmake
This page is your need.

"c++_static" is default value.
You can try "c++shared" if you like.

xtrawurst, 2019-05-09:
@RealSenseCustomerSupport 
Thanks. I have now built librealsense2.so using CMake with the Android toolchain and have selected the right STL. However, regardless of the STL we are using the resulting librealsense2.so file size is around ~100 MB (!!), even if we turn off all the optional things in the CMake configuration.

For comparison, the .so files contained in the librealsense AAR file are around ~6 MB each.

Where does this difference come from?

xtrawurst, 2019-05-09:
@RealSenseCustomerSupport 
Ok, I got to the bottom of this too:

1. The CMAKE_BUILD_TYPE must be specified on the command line. If not, then a debug build is chosen.

2. The resulting .so file still contains debug symbols that must be manually stripped. When using the Android toolchain there should be a `CMAKE_STRIP` variable available that points to the strip tool. Just call it and pass the librealsense2.so as argument.

For me the result is then a 5.6 MB library.

closed 2019-06-06 by RealSenseCustomerSupport