realsenseai/librealsense #3939: T265 fisheye stream in Unity?

FriskyBishop, 2019-05-09:
is it possible to get the T265 fisheye stream in Unity?
if not will this be added in future updates?

I assume its possible as the fisheye is now exposed
(https://github.com/IntelRealSense/librealsense/issues/3379, https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-basic)

ogoshen, 2019-05-13:
Possible, setup stream indices 1&2 with format: Y8, width: 848, height: 800 and fps: 30. 

[image]

[image]

FriskyBishop, 2019-06-20:
@ogoshen confirmed working!
thanks

closed 2019-07-31 by RealSenseCustomerSupport

johnny-wang16, 2019-10-11:
hello, I am new to unity and am still having trouble getting T265 to stream in Unity. I basically followed the instructions above but nothing shows up when I pressed play. The T265 is connected to my laptop (running ubuntu 16) through USB3.0 and I was able to see the live stream through realsense-viewer.

What I did was I created a new project, imported the sdk, instantiated the RsDevice from the "Prefabs folder", added the "Rs Stream Texture Renderer" script into the RsDevice object and configured the object as indicated above. Did I do something wrong? Any advice will be greatly appreciated!

masterchop, 2020-06-07:
This solution does not work for me, is there any unity requirement?
when I do this i get :

```
ExternalException: rs2_pipeline_start_with_config(pipe:0000000044BF4D40, config:0000000044BF5760)
Rethrow as Exception: No device connected
```
if I just do pose everything works just fine.