realsenseai/librealsense #3940: Invoked with: <pyrealsense2.pyrealsense2.composite_frame object at 0x7f4b39f5b68

bingws, 2019-05-09:
I want to cut out the region of interest , and then export it to ply file, but it report the following error.

`Traceback (most recent call last):
  File "face_detect.py", line 59, in <module>
    points.export_to_ply("2.ply", frames[x:y, w:h])
TypeError: __getitem__(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.composite_frame, arg0: int) -> pyrealsense2.pyrealsense2.frame
    2. (self: pyrealsense2.pyrealsense2.composite_frame, arg0: int) -> pyrealsense2.pyrealsense2.frame

Invoked with: <pyrealsense2.pyrealsense2.composite_frame object at 0x7f4b39f5b688>, (slice(277, 211, None), slice(427, 360, None))`

my code like following:

`import pyrealsense2 as rs
import numpy as np
import cv2
import dlib
import os

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Declare pointcloud object, for calculating pointclouds and texture mappings
pc = rs.pointcloud()

# Start streaming
pipeline.start(config)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue
        
        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        # Detect face
        frame_new = cv2.cvtColor(color_image, cv2.COLOR_BGR2RGB)
        hogFaceDetector = dlib.get_frontal_face_detector()
        faceRects = hogFaceDetector(frame_new)
        for det in faceRects :
            x = det.left()
            y = det.top()
            w = det.right()
            h = det.bottom()
            # [det.left():det.top(),det.right():det.bottom()]
            
            # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

            # Stack both images horizontally
            images = np.hstack((color_image, depth_colormap))

            # Show images
            cv2.namedWindow('RealSense')
            cv2.rectangle(images,(x,y),(w, h),(255,0,0),2)
            cv2.imshow('RealSense', images)

            # Tell pointcloud object to map to this color frame
            pc.map_to(color_frame)
            # Generate the pointcloud and texture mappings
            points = pc.calculate(depth_frame)
            points.export_to_ply("2.ply", frames[x:y, w:h])
        cv2.waitKey(1)

finally:

    # Stop streaming
    pipeline.stop()`

dorodnic, 2019-05-12:
Hi @bingws 
frames[x:y, w:h] should probably be color_frame

bingws, 2019-05-12:
Hi @dorodnic 
how to modify ? depth_frame [x:y, w:h] ?

bingws, 2019-05-14:
@ogoshen  can you help me ? thanks

ogoshen, 2019-05-15:
Regrading the error, the second param to `export_to_ply` should be a `video_frame` which is the mapped texture.

Now for the issue itself, exporting the cropped data could be a bit tricky, please consider using the [open3d](http://www.open3d.org/docs/tutorial/Basic/pointcloud.html#crop-point-cloud) library.

Using just librealsense, you could probably write a custom `processing_block` that will set the depth to zero outside of ROI, then pass it through normal point cloud calculation and export.

Notice that you'll have to either reproject the ROIs to the depth image, or work on a depth image that's been aligned to color.

bingws, 2019-05-15:
Hi @ogoshen , I had tried to export by using open3D, but get the following .ply file, the file don't contain the element face, but I need the element information. The sencond file is generated by SR300, it contain the element face.
[image]

[image]

dorodnic, 2019-06-20:
We provide single reference export format
Please look at [ply format](https://en.wikipedia.org/wiki/PLY_(file_format)) if you need to save data in a different way

closed 2019-06-20 by dorodnic