realsenseai/librealsense #3941: load custom preset in pyhton

ColinSmith123, 2019-05-09:
Hello I need some help 😄 

I try to use my custom made (with RS Viewer) json-file in pyhton. 

``
import pyrealsense2 as rs
import numpy as np
import cv2
import imageio

import json

DS5_product_ids = ["0AD1", "0AD2", "0AD3", "0AD4", "0AD5", "0AF6", "0AFE", "0AFF", "0B00", "0B01", "0B03", "0B07","0B3A"]
def find_device_that_supports_advanced_mode() :
    ctx = rs.context()
    ds5_dev = rs.device()
    devices = ctx.query_devices();
    for dev in devices:
        if dev.supports(rs.camera_info.product_id) and str(dev.get_info(rs.camera_info.product_id)) in DS5_product_ids:
            if dev.supports(rs.camera_info.name):
                print("Found device that supports advanced mode:", dev.get_info(rs.camera_info.name))
            return dev
    raise Exception("No device that supports advanced mode was found")

dev = find_device_that_supports_advanced_mode()
advnc_mode = rs.rs400_advanced_mode(dev)
print(advnc_mode)
print("Advanced mode is", "enabled" if advnc_mode.is_enabled() else "disabled")

# Loop until we successfully enable advanced mode
while not advnc_mode.is_enabled():
    print("Trying to enable advanced mode...")
    advnc_mode.toggle_advanced_mode(True)
    # At this point the device will disconnect and re-connect.
    print("Sleeping for 5 seconds...")
    time.sleep(5)
    # The 'dev' object will become invalid and we need to initialize it again
    dev = find_device_that_supports_advanced_mode()
    advnc_mode = rs.rs400_advanced_mode(dev)
    print("Advanced mode is", "enabled" if advnc_mode.is_enabled() else "disabled")

#print("Depth Control: \n", advnc_mode.get_depth_control())
#print("RSM: \n", advnc_mode.get_rsm())
#print("RAU Support Vector Control: \n", advnc_mode.get_rau_support_vector_control())
#print("Color Control: \n", advnc_mode.get_color_control())
#print("RAU Thresholds Control: \n", advnc_mode.get_rau_thresholds_control())
#print("SLO Color Thresholds Control: \n", advnc_mode.get_slo_color_thresholds_control())
#print("SLO Penalty Control: \n", advnc_mode.get_slo_penalty_control())
#print("HDAD: \n", advnc_mode.get_hdad())
#print("Color Correction: \n", advnc_mode.get_color_correction())
#print("Depth Table: \n", advnc_mode.get_depth_table())
#print("Auto Exposure Control: \n", advnc_mode.get_ae_control())
#print("Census: \n", advnc_mode.get_census())

#serialized_string = advnc_mode.serialize_json()
#print("Controls as JSON: \n", serialized_string)
    
with open("as_json_object.json", "r") as read_file:
    data = json.load(read_file)
    
as_json_object = data

# We can also load controls from a json string
# For Python 2, the values in 'as_json_object' dict need to be converted from unicode object to utf-8
if type(next(iter(as_json_object))) != str:
    as_json_object = {k.encode('utf-8'): v.encode("utf-8") for k, v in as_json_object.items()}
# The C++ JSON parser requires double-quotes for the json object so we need
# to replace the single quote of the pythonic json to double-quotes
json_string = str(as_json_object).replace("'", '\"')
advnc_mode.load_json(json_string)

    

pipeline = rs.pipeline()

config = rs.config()
#config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)   
#config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)   
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)

depth_sensor = profile.get_device().first_depth_sensor()

depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

align_to = rs.stream.color
align = rs.align(align_to)

for i in range(1000000):
        frames = pipeline.wait_for_frames()
       
        aligned_frames = align.process(frames)       
        aligned_depth_frame = aligned_frames.get_depth_frame() 
        color_frame = aligned_frames.get_color_frame()
        
        if not aligned_depth_frame or not color_frame:
            continue
        
        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((color_image, depth_colormap))
        
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        key = cv2.waitKey(1)
        
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
            
#        cv2.imwrite("../data/003/depth%d.jpg" % i, depth_colormap)
#        cv2.imwrite("../data/003/rgb%d.jpg" % i, color_image)
#               
else:       
    cv2.destroyAllWindows()
    pipeline.stop()

``

what is needed to import the jason and use it during the dataaquisition?

please I need help

MartyG-RealSense, 2019-05-09:
I hope that the Python script in the link below will be useful to you.

https://github.com/IntelRealSense/librealsense/issues/856#issuecomment-347828788

ColinSmith123, 2019-05-09:
Hey thanks for your reply - I allready tried to implement this "python-rs400-advanced-mode-example.py"  
in the code printed above.

unfortunately without success...

MartyG-RealSense, 2019-05-09:
Have you also seen Dorodnic's modification of **opencv_viewer_example.py** to cycle through presets, please?

https://github.com/IntelRealSense/librealsense/issues/856#issuecomment-347828788

dorodnic, 2019-05-12:
Hi @ColinSmith123 
So what is happening instead of expected behavior? I don't see anything immediately wrong with the code and can't run it right now

ColinSmith123, 2019-05-13:
Thanks for the replies. 

@dorodnic I don't get what you try to say with 

> In addition, load_json should also work. After reviewing your code, I had two ideas:

    Instead of using find_device_that_supports_advanced_mode, get the device from pipeline configuration after start:

cfg = pipeline.start(config)
dev = cfg.get_device()

    Perform load_json after you start streaming

> 

could you give me a specific axcample? 

Thank you very much!! :)

ColinSmith123, 2019-05-13:
I don't know how but it worked :D 

but now I have another problem... the RGB image has changed colours (see Images) 

does someone know why? 

[image]

[image]

RealSenseCustomerSupport, 2019-07-25:
----------------------------------------------

Hi @ColinSmith123,

Is the issue clarified and resolved?
As for RGB image, I wonder if the image display format via OpenCV is different with the config setting?

RealSenseCustomerSupport, 2019-08-07:
----------------------------------------------

Hi @ColinSmith123,

Any status update?

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @ColinSmith123,

I will close this ticket if no additional question. Feel free to reopen if you still have the trouble.

closed 2019-11-01 by RealSenseCustomerSupport