realsenseai/librealsense #3943: Unity SDK Documentation

TaynaCT, 2019-05-09:
So I'm tring to build a project that replaces in real time the video background of the image. But there is not enough info about the unity sdk that I can use. I'm trying to use the depth image, to detect what is close but, but it would help if I could access the distant data to optimize the final image. 

Can anyone help me?

 I Also looked at wpf example but is deprecated and can't  test

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   win 10  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.0}                          | 
| Language                            |  c# unity }                          | 
| Segment			| PC}                          | 

### Issue Description
<SDK Documentation>

MartyG-RealSense, 2019-05-09:
Could you tell us please whether you are trying to remove the background from the image (background segmentation) or substitute the background for a different image (green-screening).

If you need to remove the background, the Unity wrapper has an example program called **BackgroundSegmentation**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scenes/Samples/BackgroundSegmentation.unity

TaynaCT, 2019-05-09:
I'm trying to substitute the background for a different image (green-screening). I saw the background segmentation scene, but it doesn't give results I'm looking for.

MartyG-RealSense, 2019-05-09:
I do not personally know of a script for green-screening with the 400 Series cameras.  There is a screen-screening commercial product though that works with D415 and D435 called WildKey.  It creates green-screening with depth data instead of the traditional chromakey technique.  

It may be an option for you perhaps if you were just using Unity solely to build a green-screening app and weren't trying to integrate it as part of a larger project.

https://mixcast.me/ditching-green-screen-mixcast-gdc-2018

https://mixcast.me/docs/using-mixcast/?section=using-depth-sensing-cameras-5

TaynaCT, 2019-05-09:
ok, thank you. I'll take a look at it.

closed 2019-05-14 by TaynaCT