realsenseai/librealsense #3944: Feature Request: Better ARM (Jetson) Support

atyshka, 2019-05-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Any | 
| Firmware Version                   | Latest | 
| Operating System & Version |   Linux | 
| Kernel Version (Linux Only)    |  4.9                                       | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.20.0                         | 
| Language                            |  Any                        | 
| Segment			|  Robotics                         | 

### Issue Description
This is a general feature request for official support for platforms like NVIDIA Jetson. Intel's offerings in the robotics area are somewhat limited (NUCs have no NVIDIA gpu support) and lower power, deep learning focused boards are becoming more and more popular. Work has been done such as CUDA support which is greatly appreciated, but kernel patches remain an issue. Contributors such as JetsonHacks have made some kernel patches available, but these are almost all outdated. A quick search in the issues shows Jetson users are a large audience in the Realsense but it still feels like a subpar experience. I understand that these are not Intel devices, but it's still a big market that Intel could take advantage of.

eric-schleicher, 2019-05-10:
Similar to #3854.   the request was noted as valid.

dorodnic, 2019-12-10:
It is not perfect, but concrete steps have been taken to improve Jetson support (documentation, IMU support, firmware update, on-chip calibration and debian packages)

closed 2019-12-10 by dorodnic