realsenseai/librealsense #3945: Throttle framerate in Unity wrapper

akashcastelino, 2019-05-09:
Camera Model   -> D435
Operating System & Version -> Win10                                           
Language -> unity C#                         
         
Issue:

I am trying to use multiple Realsense D435 cameras in Unity to display their pointclouds.
When I use 4 or more cameras , Unity crashes.

So I think throttling the framerate might help while using 4 or more cameras.

Is there a way that the camera framerate can be throttled with the Unity wrapper of the Realsense SDK ?

dorodnic, 2019-05-12:
You can definitely reduce FPS or resolution. I'm not sure how to do this in Unity, @ogoshen @matkatz, but search for `EnableStream` call in the wrapper.

ogoshen, 2019-05-20:
Any other info or crash logs you can share?
Could be a memory issue, so reducing resolution rather than fps might help...

RealSenseCustomerSupport, 2019-06-20:
----------------------------------------------

Hi @akashcastelino,

Any update? Do you have a try per @ogoshen suggestion and based on the latest librealsense?

RealSenseCustomerSupport, 2019-07-04:
----------------------------------------------

Is the issue clarified? I will close this if no further question.

closed 2019-09-20 by RealSenseCustomerSupport