realsenseai/librealsense #3946: RealSense Issue with Odroid USB

JoelAdams001, 2019-05-09:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu) | 
| Kernel Version (Linux Only)    |  (18.04.2 LTS) | 
| Platform                                 | Odroid-XU4 |
| SDK Version                            |  { 2.0 }   | 
| Language                            |  {C }   | 
| Segment			|  {Robot } | 

### Issue Description
I installed the librealsense2 SDK using the Ubuntu installation guide and following the exceptions for Odroid installation very carefully. The patch I ran was the patch-realsense-ubuntu-odroid-xu4-4.14.sh and everything compiled nicely. There is a ROS node i'm running, but I get the same errors and behavior even when I open the realsense viewer.

When I plug my device into the USB 3.0 ports, I get this:
(ds5-factory.cpp:678) DS5 group_devices is empty.
and nothing works, not the depth sensor or anything.

However, when I plug the realsense device into the USB 2.1 port the depth sensor can be viewed (or published as topic on ROS) but the RGB camera doesn't work and the accelerometer and gyro doesn't work either. I do not get the DS5 group_devices is empty message in this USB port.
What further confuses me is when I run a ROS node called usb_cam, I can publish the RGB camera info perfectly fine which tells me the camera does work in the USB 3.0 port. I have looked at the other thread that talks about the group devices empty error but it doesn't give me any resolution. I will really appreciate any help.

Here is a snippet of when I run dmesg regardless of which USB port I plug into.
[image]

dorodnic, 2019-05-12:
Hi @JoelAdams001 
Should be recoverable. Is the dmesg from USB 3 or 2 connection?
What is the output of `lsusb`?
@ev-mp @RealSenseCustomerSupport - please help

JoelAdams001, 2019-05-13:
the dmesg is the same for both USB 3 and 2 connection, in the screenshot case I believe it is USB 3. Below is the output for lsusb.
[image]

RealSenseCustomerSupport, 2019-05-15:
----------------------------------------------

@JoelAdams001 What's the FW version when you get the issue?
And could you please provide the output of "lsusb |grep 8086" and "lsusb -t" when connect to USB3? Thanks!

JoelAdams001, 2019-05-15:
Thank you for coming to help. Is that referring to the firmware version? It is 2.21.0
Here are the outputs:
[image]

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

@JoelAdams001 Hi, 2.21.0 is for SDK(librealsense) version. You can check the FW(firmware) version with RealSenseViewer or rs-enumerate which should start with "05." such as 05.10.13.00.

From the output of lsusb you provided, USB3 connection can be confirmed.

Seems your issue is similar to https://github.com/IntelRealSense/librealsense/issues/3244   Please try the patch in
https://github.com/freemanlo/librealsense/blob/odroid_build_4.14/scripts/patch-realsense-ubuntu-odroid-xu4-4.14.sh

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@JoelAdams001 Did you get chance to try  https://github.com/freemanlo/librealsense/blob/odroid_build_4.14/scripts/patch-realsense-ubuntu-odroid-xu4-4.14.sh  Looking forward to your update. Thanks!

JoelAdams001, 2019-05-21:
@RealSenseCustomerSupport I apologize for the delay. The firmware version is 05.10.13.00, and I just ran the patch and still have the issue.

Also, the rs-enumerate-devices command doesn't detect the device in USB 3 interestingly.
[image]

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@JoelAdams001 When you connect the camera to USB3, can you still get previous output for "lsusb |grep 8086" and "lsusb -t"? From your previous output, the camera can be detected when connect to USB3. Do your device have several USB3 ports? Does all the USB3 ports have the same result?

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@JoelAdams001 Did you get it through?

JoelAdams001, 2019-06-13:
@RealSenseCustomerSupport Sorry for the delay. There are multiple USB3 ports, two of them exactly. I believe the outputs are the same. I'm going to get the data but I'm having a separate issue using SSH on it at the moment, will try getting this tomorrow

JoelAdams001, 2019-06-19:
Sorry again, i've moved on from the odroid in my project.

closed 2019-06-19 by JoelAdams001