realsenseai/librealsense #3947: Generate a Pipeline Profile from a SoftwareDevice

stephen-systemfriend, 2019-05-10:
I would like to (in C#) create a pipeline profile for my Software device so I can use the native processing blocks on the output from the software device.
Right now the PointCloudRenderer chokes up because the ActiveProfile sent in OnStart is null because I can't initialise one.

dorodnic, 2019-05-12:
Hi @stephen-systemfriend 
To get PipelineProfile from SoftwareDevice, the way I know is to add the device to Context and then create Pipeline from that Context, and make sure it selects the software device. I'm not 100% sure if adding software device to Context is wrapped in C#, but should be simple. 
In general, we are still figuring out some of the flows involving SoftwareDevice. In theory, this should be extremely useful for combining RealSense with external devices and 3rd party data formats. Feedback on how customers are using this feature will help us improve it.

stephen-systemfriend, 2019-05-14:
Thanks, created a PR for the fix

stephen-systemfriend, 2019-05-14:
how do I specify that it selects the software device over the ones it auto detects? (I'd also be happy to have it not automatically detect other devices, because it becomes very messy when I have multiple processes trying to create contexts with a T265 plugged in, though this is just for development)

ogoshen, 2019-05-20:
Serial number filtering on the `rs2::config` is usually how you select specific devices.
Unfortunately software devices don't support setting their info (`RS2_CAMERA_INFO_SERIAL_NUMBER`).

You might get away with adding a dummy stream to your software device, just as to mark it for `config::enable_stream` to pick it up...

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

Since this PR created by @stephen-systemfriend was merged since librealsense 2.22.0, is there any additional effort for this ticket? If not, will close this.

closed 2019-09-20 by RealSenseCustomerSupport