realsenseai/librealsense #3948: get distace using D435

scizors, 2019-05-10:
Hello,
 I am using pyrealsense2 in python 3.6. I would loke to know how can I measure the ditance between the 2 points A and B in matplotlib using intel realsence D435

MartyG-RealSense, 2019-05-10:
A RealSense user called soarwing32 who was also using matplotlib and D435 with Python wrote their own conversions of the SDK's 'Measure' C++ example.  One of these scripts is titled 'measure_with_mpl.py'

https://github.com/soarwing52/RealsensePython/blob/master/drafts/measure_with_mpl.py

soarwing's other Measure conversions for Python can be found in the Drafts folder on their GitHub.

https://github.com/soarwing52/RealsensePython/tree/master/drafts

soarwing52, 2019-05-10:
@MartyG-RealSense  thank you for the mention
I used several methods, first openCV with two mouse click events, but it the visualization was not ideal

then I looked into matplotlib widgets, there is a Ruler, I added the 3D transformation into it

the wait_for_frame matching is not perfect so I wrote again get two frames timestamps gap within 20 ms
and loop through, it takes longer but accurat, I wonder if there will be a better way

and also the set_real_time(False) freezing issue is still there, so I used subprocess to call it

MartyG-RealSense, 2019-05-10:
@soarwing52  Thanks so much for your quick and helpful response!

scizors, 2019-05-10:
how can I measure the ditance between the 2 points A and B

soarwing52, 2019-05-10:
tip: look into the rs.rs2_deproject_pixel_to_point funtion

scizors, 2019-05-10:
runfile('C:/Users/rsr5le/Desktop/realsense/print.py', wdir='C:/Users/rsr5le/Desktop/realsense')
Traceback (most recent call last):

  File "C:\Users\rsr5le\.conda\envs\python3.6\lib\site-packages\IPython\core\interactiveshell.py", line 3296, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)

  File "<ipython-input-9-4104b76289f0>", line 1, in <module>
    runfile('C:/Users/rsr5le/Desktop/realsense/print.py', wdir='C:/Users/rsr5le/Desktop/realsense')

  File "C:\Users\rsr5le\.conda\envs\python3.6\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Users\rsr5le\.conda\envs\python3.6\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/rsr5le/Desktop/realsense/print.py", line 19
    print file_name
                  ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(file_name)?

scizors, 2019-05-10:
Thanks for the reply .i am having error in this file .I need it some realtime viewer in which I can measure the depth distance .

scizors, 2019-05-10:
file_name = '0327.bag'

could you say what is this and where is this file

soarwing52, 2019-05-10:
so, with my project I have my images recorded, which is the .bag file, that's why
if you want to do real time measurements, you need to adjust the codes,
and I'm using python2.7 because my project is combining with ArcGIS 10.4, you would also need to adjust which like the print function will require parentheses
the calculations are in the widgets2.py, the Ruler tool

scizors, 2019-05-10:
Thanks a lot for your help @soarwing52

closed 2019-05-10 by scizors

jaimeet64, 2021-06-07:
> https://github.com/soarwing52/RealsensePython/blob/master/drafts/measure_with_mpl.py
This link doesn't work. Do you (@MartyG-RealSense ) know any example script of rs-measure in python?

MartyG-RealSense, 2021-06-07:
@jaimeet64  Soarwing52's Python conversion of rs-measure is now located at the link below, as he expanded it into a larger measuring project but still kept its individual functions as standalone scripts.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py