realsenseai/librealsense #3949: color texture mapping doesn't work (at least in RealSense Viewer)

kartur, 2019-05-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  -                                       | 
| Platform                                 | x86-64 with Nvidia Quadro M1200 (419.67)|

### Issue Description
Realsense Viewer can't map the color into 3d space. It's just mapping some grey flickering. Grabbing all the different sources itself works though. See screenshots.
[image]
[image]
[image]

dorodnic, 2019-05-12:
Hi @kartur 
There is a known issue right now of D435i losing Depth to RGB calibration. The team identified root cause and we will roll out a fix (most likely this week).
Could you please attach the results of `rs-enumerate-devices -c` to make sure this is the issue.

kartur, 2019-05-20:
sure
[output_rs_enu_dev_c.txt](https://github.com/IntelRealSense/librealsense/files/3198246/output_rs_enu_dev_c.txt)

kartur, 2019-05-20:
I just tried the release from today, and it fixes the issue. Thanks. :)

ev-mp, 2019-05-20:
@kartur ,thanks for the confirmation.
Can it be closed now ?
#3788, #3201, #3474,

taletovic, 2019-05-21:
Hello, I am facing unfortunately the same problem, even with the new v2.22.0 on Ubuntu 16.04.

Here is the output of the `rs-enumerate-devices -c`:

[rs-enumerate-devices-result.txt](https://github.com/IntelRealSense/librealsense/files/3201041/rs-enumerate-devices-result.txt)

The 3D color to depth is just black:

[image]

DynamicCalibrator couldn't fix the problem:
[image]

I've tried `Intel.Realsense.CustomRW -g` too.

Is there any new firmware to update or what should I do?

Thanks for your support.

ev-mp, 2019-05-21:
@taletovic , hello.
The error you encounter is not the same as for the first person.
>Extrinsic from "Depth"	  To	  "Color" :
N/A

Please elaborate when  you first encountered this and what operations preceded/followed. (FW updgrades/downgrades)

In any case you can try  `realsense-veiwer` **v2.22** from the latest release so see whether this helps.
Then generated and shared the log of `rs-enumerate-devices -c`

taletovic, 2019-05-21:
@ev-mp Thank you for your quick answer.

I got my D435i a month ago. I'm not 100% sure but I think the FW version was 5.11.4.
Than I tried to update to the newest version **5.11.6.200** and noticed that the depth to color mapping doesn't work.

After that I tried downgrading to **5.11.4** which didn't help.
Meanwhile I saw that you are working on the solution so I upgraded to **5.11.6.200** again.

I can say one for sure, on the **v2.21.0**, I had exactly the same problem as @kartur. The feed in the `realsense-viewer` was grey like in his post.

I was following your posts and saw yesterday that there is a new release **v2.22.0** so I tried it.
The results are in my last post.

taletovic, 2019-05-21:
@ev-mp sorry for double post.
I've just tested that what I already said. Since I had the **v2.21** on my Windows PC I've tested the `rs-enumerate-devices -c`. The result is:

[rs-enumerate-devices-win10_v2.21.txt](https://github.com/IntelRealSense/librealsense/files/3201364/rs-enumerate-devices-win10_v2.21.txt)

I tested with two different versions of RealSenseViewer and here are the results:
[image]

[image]

I had the same situation before updating to **v2.22** on my Ubuntu 16.04.

ev-mp, 2019-05-21:
@taletovic hello, can you provide the output of the `rs-fw-logger` as well ?  - 
- Close all realsense applications
- Run `rs-fw-logger > log.txt`
- Run `rs-enumerate-devices -c`
- Post the log.txt for review
Thank you for cooperation

taletovic, 2019-05-21:
@ev-mp sure.

I was logging about a minute long, hope it is enough data to analyse:

[log.txt](https://github.com/IntelRealSense/librealsense/files/3202183/log.txt)

And again `rs-enumerate-devices -c` (using **Ubuntu 16.04** and **v2.22.0**):

[rs-enumerate-devices_output.txt](https://github.com/IntelRealSense/librealsense/files/3202185/rs-enumerate-devices_output.txt)

kartur, 2019-05-21:
> @kartur ,thanks for the confirmation.
> Can it be closed now ?
> #3788, #3201, #3474,

I can't test if it also works when using the library, since with the new version, the pipeline doesn't work for me anymore: https://github.com/IntelRealSense/librealsense/issues/4028

Within the realsenseviewer it's def. fixed for me. 
But I think it would be best to confirm it also works outside the viewer again, before we close this ticket. :)

taletovic, 2019-05-21:
@kartur Hello,
I see that you are using the Windows 10 version of the SDK.

How did you run the Realsense Viewer, installed completely the new **SDK v2.22.0** or you launched the standalone **Intel.RealSense.Viewer.exe**?

If you run only the **Intel.RealSense.Viewer.exe**, I think you have to completely uninstall/remove the old **SDK v2.21.0** and then install the newest release:

[https://github.com/IntelRealSense/librealsense/releases/download/v2.22.0/Intel.RealSense.SDK-WIN10-2.22.0.856.exe](url)

kartur, 2019-05-21:
@taletovic Hi,
I first tried the standalone. But I also already tried the version that comes with the SDK (2.22.0) and that also worked fine.

Do I understand correctly, that you recommend not installing new versions of the SDK over the old one?

taletovic, 2019-05-22:
@kartur Hi,
like in my post:
[https://github.com/IntelRealSense/librealsense/issues/3949#issuecomment-494285059](url)

You see that I had different problems with two different versions of the SDK on my Windows PC. I would suggest you to reinstall the newest SDK and then try the Visual Studio Examples.

kartur, 2019-05-22:
@taletovic Hey, thanks for the hint. But unfourtantely it didn't change anything for me.

Anyways, since my original issue is fixed and I think you fixed yours too, I will close this issue, ok?

taletovic, 2019-05-23:
@kartur Hello
No, my problem is still not fixed, as you can see I am getting different results with `rs-enumerate-devices -c` on different versions and none of them works (grey or black image).

ev-mp, 2019-05-23:
@taletovic hello,
Please create and share the camera dump generated with the attached python script 
[realsense_flash_debug.txt](https://github.com/IntelRealSense/librealsense/files/3211870/realsense_flash_debug.txt)

Instructions:
1. Install python 2.7/3.6
2. Install realsense python wrapper
`pip install pyrealsense2`
3. Rename the attached script extension: `realsense_flash_debug.txt` into `realsense_flash_debug.py`
4. Run the script
`python realsense_flash_debug.py`
The script will run for ~2min and generate device FW dump `realsense_debug_dump.bin`
4. Attach the resulted file

taletovic, 2019-05-23:
@ev-mp Hello,

my zipped `realsense_debug_dump.bin` file:

[realsense_debug_dump.zip](https://github.com/IntelRealSense/librealsense/files/3211916/realsense_debug_dump.zip)

Hope this helps. Thank you.

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Hello,

Just wonder, is this issue resolved on your side by the latest firmware and RS Viewer?

Thanks

RealSenseCustomerSupport, 2020-01-31:
----------------------------------------------

Hi @taletovic Do you still have question on this?
If we don't hear from you in 7 days, we will close this issue.

closed 2020-02-07 by RealSenseCustomerSupport