realsenseai/librealsense #3952: node-librealsense not working on macOS 10.13.6

leoregner, 2019-05-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |  macOS High Sierra 10.13.6  | 
| Kernel Version (Linux Only)    |  ---                                         | 
| Platform                                 | MacBook Pro 15 inch, late 2011  |
| SDK Version                            | 2.0                      | 
| Language                            | NodeJS                        | 

### Issue Description
I've been trying to get started with my RealSense camera in NodeJS. The installation of the library appeared to run smoothly and without any errors ...

```bash
mkdir test && cd test && npm init -y
npm i node-librealsense
cd node_modules/node-librealsense && npm i
cd examples && npm i
```

... however, none of the provided examples seems to work for me:

### nodejs-align.js
```bash
examples leo$ node nodejs-align.js 
Press Up/Down to change the depth clipping distance.
/Users/leo/Desktop/test/node_modules/node-librealsense/examples/nodejs-align.js:87
  let colorFrame = alignedFrameset.colorFrame;
                                   ^

TypeError: Cannot read property 'colorFrame' of undefined
    at Object.<anonymous> (/Users/leo/Desktop/test/node_modules/node-librealsense/examples/nodejs-align.js:87:36)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:754:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)
```

### nodejs-capture.js
```bash
examples leo$ node nodejs-capture.js
/Users/leo/Desktop/test/node_modules/node-librealsense/examples/nodejs-capture.js:35
        color.data, 'rgb8', color.width, color.height);
              ^

TypeError: Cannot read property 'data' of undefined
    at Object.<anonymous> (/Users/leo/Desktop/test/node_modules/node-librealsense/examples/nodejs-capture.js:35:15)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:754:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)
```

### nodejs-pointcloud.js
```bash
examples leo$ node nodejs-pointcloud.js
Drag to change perspective, scroll mouse wheel to zoom in/out.
/Users/leo/Desktop/test/node_modules/node-librealsense/index.js:3380
          throw new TypeError(wrongTypeErrMsgPrefix + expectedType.name);
          ^

TypeError: argument 0 of PointCloud.mapTo() should be a/an Frame
    at checkArgumentType (/Users/leo/Desktop/test/node_modules/node-librealsense/index.js:3380:17)
    at PointCloud.mapTo (/Users/leo/Desktop/test/node_modules/node-librealsense/index.js:1733:5)
    at Object.<anonymous> (/Users/leo/Desktop/test/node_modules/node-librealsense/examples/nodejs-pointcloud.js:42:6)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:754:12)
    at startup (internal/bootstrap/node.js:283:19)
```

### realsense_viewer
```bash
realsense_viewer leo$ node index.js 
unsupported descriptor subtype: 3
unsupported descriptor subtype: 3
unsupported descriptor subtype: 3
unsupported descriptor subtype: 3
listening on http://localhost:3000
Command connection established!
Segmentation fault: 11
```

### nodejs-save-to-disk-js
Only the `save to disk` script runs through without throwing up, but the resulting image files are disturbed and thus pretty much unusable.

```bash
examples leo$ node nodejs-save-to-disk.js
Saved  rs-save-to-disk-output-depth.png
Saved  rs-save-to-disk-output-infrared.png
```

[image]

[image]

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

@leoregner Which librealsense version did you get the issue? And noticed the camera firmware you used is v5.9.2 which is a little old, could you please update to the latest librealsense and FW to see if you still get the issue? Thanks!

leoregner, 2019-06-24:
> @leoregner Which librealsense version did you get the issue? And noticed the camera firmware you used is v5.9.2 which is a little old, could you please update to the latest librealsense and FW to see if you still get the issue? Thanks!

@RealSenseCustomerSupport Thanks, this solved the problem. Anyways, I had already moved to Linux in the meantime.

closed 2019-06-24 by leoregner