realsenseai/librealsense #3953: The Viewer and the Depth Q.T.  are not working again

TDVL, 2019-05-10:
Dear Support Team,

I have a problem again. The Viewer and the Depth Q.T.  are not working again. They didn't start. The rs-capture works. The SDK version is 2.21.0.

The old version 2.17.1 works only, like in my previous post about this. I have updated the Windows to newer version 1709. According to your specifications, all the SDK versions should work on this Windows. All the other programs work perfectly.

What wrong with the SDK?

Thank you in advance.

Laslo

TDVL, 2019-05-11:
Hello,
I haven't noticed yesterday, the Record option is not working in 2.17.01. version, but I can see the RGB and Depth streams. Also, I can save them in PNG, but I can't save the BAG file with Record button. When I press the Record button, the RealSense Viewer closes. This is all happening in 2.17.1. The newer versions of the Viewer and Depth Q.T. I can not start.
Thank you.
Best regards

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

 Hi TDVL,

If it works before, maybe there were something else changed outside RS tool suite.
I have 2.17.1, 2.19.0, 2.21.0 all working well here (DQT and RS Viewer) with Windows10 1709.

Can you please either download again or try build from source? Either way has no issue for my Windows10 1709.

Thanks!

Attachment(s):
RS_DQT_2.17.1_Win10-1709.png - https://realsensesupport.intel.com/attachments/token/egpRj8uLFtmyXOgVoMLGvSwyb/?name=RS_DQT_2.17.1_Win10-1709.png
RS_DQT_2.21.0_Win10-1709.png - https://realsensesupport.intel.com/attachments/token/F2sS49ehcTTdNJs9DDWHXbtYn/?name=RS_DQT_2.21.0_Win10-1709.png

RealSenseCustomerSupport, 2019-05-28:
----------------------------------------------

Hi TDVL,

Do you get any update on this one? Or do you still need help?

Thanks!

closed 2019-06-03 by dorodnic