realsenseai/librealsense #3954: Unable to Set Frame Rate on RealSense D435

cookem4, 2019-05-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 }| 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Linux (Ubuntu 16/17) | 
| Kernel Version (Linux Only)    | 4.4.38 | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  2.0.0                        | 
| Language                            |  C++   | 
| Segment			|  Robot | 

### Issue Description
Hello,

I have been having troubles setting the desired frame rate on the RealSense D435. Within the librealsense library, there is a method in the rs_config.cpp file that allows the camera mode, resolution, and frame rate to be modified, called enable_stream(). The problem I am facing is that although this method is supposed to modify the resolution and channels, it has no effect on my camera. I confirmed that it has no effect but using the function to set the frame rate to 6Hz at 848x480 and was not choppy like 6Hz should be. The following link shows an example program (using openCV) from Intel that uses the same enable_stream() method no differently than I do.

Using the realsense-viewer application, the frame rates are able to be changed , but upon writing my own C++ program I cannot change them. Additionally, I have determined that the color channel is running at 30Hz and is unable to be changed from that with C++ code and that the depth channel is running at 20Hz and is unable to be changed.

It would be greatly appreciated if anyone had information on where to go from here and what my problem might be.

Best regards,
Mitchell

dorodnic, 2019-05-12:
```
using namespace rs2;
config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_RGB8, 6);
pipeline p;
p.start(cfg);
while (p.wait_for_frames()) std::cout << ".";
```
Should be printing at 6 dots per second. 
Feel free to share your code, and please double check you are passing the config to `start`

cookem4, 2019-05-13:
This works thank you! I didn't think to pass the config object to the pipeline upon starting it.

closed 2019-05-13 by cookem4