realsenseai/librealsense #3957: C-api: Advanced mode

rowdyroad, 2019-05-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.200 | 
| Operating System & Version |  ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-48-generic                                       | 
| Platform                                 | PC |
| SDK Version                            |  2.21.0.0                       | 
| Language                            |  C }                          | 
| Segment			|  IoT                         | 

### Issue Description

Hi.

When I'm turning on advanced mode and waiting for depth frames, I've got 3-5 frames and nothing...
It's freezed until RS2_DEFAULT_TIMEOUT error raised. 

dmesg is fresh and clean.

realsense-viewer is working correctly.

But if I add some delay (about 10 seconds) after turning advanced mode on, everything is ok. Looks like advanced mode reboots camera. 
Is it expected behavoir or I do something wrong?

Also, how can I check what camera is ready to send me depth frames after advanced mode is turned on?

dorodnic, 2019-05-12:
Hi @rowdyroad 
Advanced mode flag is persistent, so doing it once per camera should be sufficient. 
However, if you want to be robust in your code, you can handle this transition in code. 
Prior to main flow, use `context().query_devices()` and check if the device is in advanced mode. If not, toggle advanced mode and wait for camera re-connection (one way is using device hub API, for C it would be `rs2_create_device_hub` and `rs2_device_hub_wait_for_device`)

dorodnic, 2019-06-03:
Hope this helped

closed 2019-06-03 by dorodnic