realsenseai/librealsense #3959: Dnn example not working properly.

Ahad19931, 2019-05-12:
Camera Model D435i
| Firmware Version 05.11.06.200
| Operating System & Version | Linux (Ubuntu 16.04 LTS)
| Kernel Version (Linux Only)| 4.15.0-48-generic
| Platform| PC
| SDK Version 2.21.0
| Language python/C
| Segment	Robot

The dnn example is not working properly. It keeps on showing the distance as '0 meters away' even if the distance of the object from the camera changes. See the screenshots to get a better idea of the problem.

[image]
[image]

dorodnic, 2019-05-12:
Hi @Ahad19931 
Please check if `rs-align` example is working properly. This might be related to known calibration issue of D435i being addressed right now

Ahad19931, 2019-05-12:
Yes, the rs-align example is working properly but the dnn example is not working properly.

Ahad19931, 2019-05-12:
Here is the screenshot for the rs-align example. Please check if the output of this example is correct.

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

Hi @Ahad19931,
Thank you for attaching the output of rs-align.
Based on the output, it seems that dorodnic was correct about running into a known calibration issue.
Could you please give SDK Version 2.22.0 a try?
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#bug-fixes

closed 2019-06-08 by RealSenseCustomerSupport