realsenseai/librealsense #3960: Imperfect Alignment of color and depth image

Bingobingo123, 2019-05-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  SR300  } | 
| Operating System & Version |   { Linux (Ubuntu /16/)  | 
| Kernel Version (Linux Only)    |  4.15.0-47-generic
                                     | 
| Platform                                 | PC  |
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hello Community, I have a encountered a problem when I tried to align the depth and color image taken from SR300.

I used the following code to do the alignment:

		# aligned frame
            	align = rs.align(rs.stream.color)
            	aligned_frames = align.process(frames)
            	color_frame = aligned_frames.get_color_frame()
		aligned_depth_frame = aligned_frames.get_depth_frame()

		# Convert images to numpy arrays
		color_image = np.asanyarray(color_frame.get_data())
		aligned_depth_image = aligned_depth_frame.get_data()
		depth_image = np.float32(aligned_depth_image)
Here is the result:
[image]

The green dot in each figure is the center of the figure.

After the alignment, the two figures are not 100% aligned to each other since the green dot is not in the center of the object in the color image but the green dot in the depth image seems to be in the center of the object.

Could anyone help? Thanks in advance!

RealSenseCustomerSupport, 2019-06-18:
----------------------------------------------

Hi @Bingobingo123,

I wonder if your SR300 has the calibration issue. Any bending effect on your SR300 peripheral?

RealSenseCustomerSupport, 2019-08-19:
----------------------------------------------

Hi @Bingobingo123,

Any update?
1. What's your requirement of depth-color alignment?
2. What's your test distance with the target and the resolution you use for depth and color? If you align depth pixel (in the center of the depth frame as green dot) to color, the aligned depth pixel in color frame is not in the center of the picture.
You can refer to below page, "rs2_deproject_pixel_to_point", "rs2_transform_point_to_point" and "rs2_project_point_to_pixel" functions can be used to align depth pixel to color.
https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

Hi @Bingobingo123,

If the issue clarified, will close this.

closed 2019-11-01 by RealSenseCustomerSupport