realsenseai/librealsense #3961: D415-why the roof is seen in depth image on the left side of the wall?

TDVL, 2019-05-12:
Hello,
The distance for visualization was set between 80-100cm, the camera was 90cm from the wall. The high accuracy preset was set, also the edge filtering was on (according to the Intel's whitepaper). The roof is about 6-7 meters far from the camera. What is wrong here? Is it possible to separate correctly the wall from the window and the background with D415 (like I have separated the shape from the wall in the image)?
[image]
[image]
[image]
[image]
Thank you in advance.
Best regards

agrunnet, 2019-05-12:
Is it possible that "yellow" in this case is infinity, or beyond 100cm?
The problem then would not be the background and roof, but why parts of the wall is showing up at much farther away. This may happen if the camera is over exposed or there are no features on the wall. 
If you look at the "left image" it will help to see whether the imagers are saturated or simply don't see texture.

agrunnet, 2019-05-12:
Same as issue #3962 I believe.  
https://github.com/IntelRealSense/librealsense/issues/3962
Please try to reduce the exposure and gain manually and see if it helps.

TDVL, 2019-05-12:
Dear Anders,
I have tried to reduce the exposure and the gain, but without significant results.
Yes, in most cases there are no features on the wall and will not be any features. 
Hence, it is not possible to extract the wall with the D415 camera reliably? The solution should be universal for all walls, to separate the wall based on the depth information from the window or from the 
ornament.

TDVL, 2019-05-16:
Dear Anders,
Accordingly, with D415 I can't solve this problem? Without a great depth image I can't extract the specified walls (not only the wall from the image). Do you have any suggestion? Maybe another camera or some other equipment?
Thank you,
Best regards.

agrunnet, 2019-05-16:
Have you tried the D435?
From your image it looks like there should be enough texture to get very good depth.

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Hi @TDVL,
Please double check and perhaps to try to adjust the depth visualization range as @agrunnet suggested:

> Is it possible that "yellow" in this case is infinity, or beyond 100cm?

It seems that all the depth points that are beyond 100cm are represented in yellow.

TDVL, 2019-05-18:
Hello Support Team,
The depth visualization range was between 0,8-1m, the camera was on 0,9m from the wall. The setting was not infinity. This is the problem. I tried to set the exposure, but without improvements.
I don't know what is the problem, the camera setting or the camera itself? 
I need the wall surface in one, unique color in depth image, or I can not detect the wall after the binarization and processing (the multicolored wall surface in depth image produces big losses from the wall and these losses later are not recoverable).
Any suggestion? Maybe you can try it outdoor on your wall and window?
Thank you.
Kind regards.

RealSenseCustomerSupport, 2019-06-06:
----------------------------------------------

Hi TDVl,
With the RealSense viewer and the stereo module stream, could you please check the depth (distance) from the wall vs the roof? Move the mouse cursor inside the depth stream display and the distance will be shown at the bottom of the

RealSenseCustomerSupport, 2019-06-06:
----------------------------------------------

Hi TDVl,
With the RealSense viewer and the stereo module stream, could you please check the depth (distance) from the wall vs the roof? You can move the mouse cursor inside the depth stream display and the distance will be shown at the bottom of the display window. Please see the attachment.
https://intel.sendsafely.com/receive/?thread=RAB1-PFUT&packageCode=gd7CVh2xt1jL7zpWN189XlWZ99Yh7BMnjSgDyyfaiAk#keyCode=FAKmI7askUsYqy33MeZQgxI2kQAL2lQ3dIx2y5jyOHs

Attachment(s):
depth_dist.png - https://realsensesupport.intel.com/attachments/token/2xfpOvnxLDn5DDL89SahJIjTg/?name=depth_dist.png

RealSenseCustomerSupport, 2019-06-27:
----------------------------------------------

Closing this issue for now. Please feel free to reopen if issue is still relevant. Thank you.

closed 2019-06-27 by RealSenseCustomerSupport