realsenseai/librealsense #3962: Is it possible to separate, extract the wall from the background (window, etc.)?

TDVL, 2019-05-12:
Hello,
I am working on a image processing project where the goal is to extract the wall from the background using depth image from the D415 camera. But, the depth information of the wall is acting like a background and after the binarization the information loss is large from the wall and it is not possible to recover it. The high accuracy preset was used with edge preserving filters in Viewer.
Is there something wrong with the camera settings and use? Are there any appropriate settings for this kind of capturing?
Thank you.
[image]
[image]
[image]

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

Hi @TDVL,

Any test update? Do you have a try of D435 also?

TDVL, 2019-10-13:
Hello,
Sorry for late response, I had some other projects to finish.
Yes, I have tried, but without success. The same problem persists.
Laslo

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

Hi @TDVL,

One example in below link. Please check if it's helpful for your use case.
https://github.com/pedropro/CAPE

closed 2019-12-03 by RealSenseCustomerSupport