realsenseai/librealsense #3964: Any Possible Way to Get Depth from T265

mzahana, 2019-05-13:
Hi.

I know that T265 is currently used as a tracking camera. Although it is a stereo camera, but no depth is provided. Is there any possibly way to get depth from T265?

MartyG-RealSense, 2019-05-13:
The official answer in terms of RealSense support for this is of course "no".  However, some researchers have used fisheye camera pairs in general (not the T265 ones specifically) to generate a depth map for applications such as self-driving cars.  Google "get depth from fisheye camera" for more details.  

Here is an example paper:

https://arxiv.org/pdf/1809.06132.pdf

ev-mp, 2019-05-13:
@mzahana,
Please check https://github.com/IntelRealSense/librealsense/pull/3951 by @bfulkers-i that demonstrates how to generate depth map with T265 fisheye stereo sensors.

MartyG-RealSense, 2019-05-13:
@ev-mp  Thanks so much for the link!   It's awesome that someone has managed to put the depth from stereo fisheye principle into practice with T265 and also deal with the fisheye distortion factor that otherwise makes depth calculation from fisheye awkward   :)

mzahana, 2019-05-13:
@ev-mp thanks for the information.

ShettyHarapanahalli, 2019-05-13:
If you are using any Nvidia product then you can try to use stereo processing algorithms like one in the link below to get depth from the stereo cameras - 

https://docs.nvidia.com/isaac/isaac/packages/stereo_depth/doc/sddnn.html 

Best Regards,
Suman

mzahana, 2019-05-13:
@ShettyHarapanahalli thanks for the information. The stereo_depth_dnn, according to my understanding, would run at 10-30 FPS on TX2 which can be power hungry for small robots. I guess a depth camera like D4xx would be better in this case. Also, if depth can be computed on T265 as you mentioned, it would probably solve the power issue and avoiding the use of an extra depth camera. However, I am not sure if the depth quality would be comparable to D4xxs.

dorodnic, 2019-05-15:
@ShettyHarapanahalli @mzahana - it is possible to do basic depth in T265 firmware, however, IMHO it would be low priority, because resulting depth quality would not be acceptable for most applications (both because of passive nature of the device and because of the wide field of view)
Optics of T265 are optimised for the task of tracking while these of the D400 - for dense depth generation. 
Even if you only need sparse depth for something like collision avoidance, without a projector there is no guaranty you would get any depth from the obstacle.

ShettyHarapanahalli, 2019-05-16:
Field of view of T265 is much better compared to D4xx and for many application, we need broader viewing angles the depth would be icing for this kind of applications.  Considering obstacle avoidance we need accurate depth just in front of the robot, other places low-quality depth will do the job.  

I believe next-generation tracking cameras will have these features - 
1. Depth is also provided 
2. Integration of GNSS and other technologies to get better localization
3. AI is deployable with the device for object detection and other applications

ShettyHarapanahalli, 2019-05-27:
Good to see the depth generation from T265 .. 
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py

RealSenseCustomerSupport, 2019-06-05:
----------------------------------------------

Hi @mzahana

With the inclusion of the python example, can we close this issue now?

Thanks

closed 2019-06-17 by RealSenseCustomerSupport

alexcherpi, 2019-06-26:
Hi @MartyG-RealSense 
You said that we could use standard algorithm such as in (https://arxiv.org/pdf/1809.06132.pdf) to reconstruct the depth map from the right&left images. 
But to my knowledge the T265 has a IR projector which projects a dot pattern both left & right cameras can see. I'm afraid that given all these dots in the image standard depth-from-stereo algorithm would lead to errors during the reconstruction. 
What do you think? Thanks

MartyG-RealSense, 2019-06-26:
Since that PDF was quoted, an example program for getting depth from the T265's left and right fisheye cameras has been added to the SDK.   It isn't as good as the depth on a dedicated depth camera such as the D415 or D435 but certainly better than nothing.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py

As far a I know, the T265 does not have an IR projector.  The purpose of the dot pattern cast by the projector in cameras such as the D415 and D435 is to add texture to objects to aid a camera in reading depth information from a surface.  But as the T265 does not have depth sensing by default and has to obtain it with a workaround on the fisheye lenses, it would not need a projector to enhance depth.

alexcherpi, 2019-06-26:
So you mean than the T265 can provide depth map on textureless surface without an IR projector whereas the D453 needs an IR projector?

MartyG-RealSense, 2019-06-26:
The D-models can operate without the dot pattern, using ambient light such as sunlight as the means to analyze depth information.   Dot projection is just a useful feature to have in environments where ambient light is not strong.  Unlike most depth cameras, the 400 Series actually performs better in strong sunlight.

agrunnet, 2019-06-26:
There are no plans to compute depth inside the T265. And you are correct, that on truly textureless surfaces it is not possible to generate good depth without a projector.

tiralonghipol, 2021-01-30:
https://github.com/tiralonghipol/t265_depth