realsenseai/librealsense #3965: How to play infrared frame from rosbag with matlab?

MrNash97, 2019-05-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.21.0                           | 
| Language                            |  Matlab                         | 
| Segment			|  others                           | 

### Issue Description
Hi,
I have recorded .bag file contain RBG and infrared (IR1 and IR2) data with my d435. I'm using rosbag_example.m to stream both, the RGB and infrared data. From that example, I was able to stream the RGB data but clueless with how to stream the infrared.

And one more thing, it seems the RGB and infrared data have different fps (I check it using realsense viewer). Is there any example of how to stream the data with the timestamp? So I can understand the data better.

Thankyou

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Hi MrNash97,

Wonder if this can help you, if not checked yet?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/rosbag_example.m

Thanks!

MrNash97, 2019-05-18:
I've tried that code. It works well with depth and RGB data. But, what I want to know is how to display the Infrared data. Can you help me with how to access the IR data? I only know how to access depth and RGB.
`depth = fs.get_depth_frame(); ` This one is for depth
and
`color = fs.get_color_frame(); ` This one is for RGB.

Thank you.

lramati, 2019-05-19:
If the bag contains IR files, they should be streaming already. When no streams are explicitly specified to the config object for a bag file, all streams present are opened. `fs.get_infrared_frame(0)` and `fs.get_infrared_frame(1)` should allow you to grab the individual frames as they are read from the bag

MrNash97, 2019-05-20:
@lramati Thank you, it's working. 
I want to ask one more thing. Is there any way to access the timestamp of each frame in Matlab? I need to synchronize the timing between IR and RGB data because it has different fps (even though I set the recording with the same fps).

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

Hi MrNash97,

You can refer to the timestamp functions defined in frame.m for how to get timestamp info:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/frame.m#L24
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/frame.m#L27

Hope this can be helpful.

Thanks!

MrNash97, 2019-05-25:
Thank you for your help. It seems that I must enable the frame metadata first to get what I wanted.

closed 2019-05-25 by MrNash97