realsenseai/librealsense #3966: Realsense can't detect grids/meshes

thanujadax, 2019-05-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.18.1  | 
| Language                            |  C++ (ROS) }                          | 
| Segment			|  Robot                          | 

### Issue Description
The realsense camera seems to have problems detecting grid (or mesh) like structures as shown in the image (pointcloud visualization using RVIZ). Some of the points which are supposed to be on the grid are detected elsewhere. After intrinsic calibration it gets slightly better, but still exists. Any ideas to overcome this?
[image]

AntoineWefit, 2019-05-14:
If you haven't read it, I would advise you to take a look at the chapter 14 of [this paper](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf).
You could also try to enable some post processing in the Viewer/your code to help closing the smaller holes and improve the overall quality of your mesh.

Hope this'll help !

dorodnic, 2019-05-14:
Hi @thanujadax 
Did you try loading High Accuracy JSON and increasing laser power to max?
The preset can help invalidate wrong pixels while getting the projector pattern to be visible in the infrared image is likely to remove the problem. 
One extra thing people had success with in this type of scenario is adding a physical filter in front of the camera that passes only IR spectrum. I know it's not exactly a household item, but still. 
The core of the problem is that stereo matching algorithm can get confused when there are many very similar features in one horizontal row. This is why rotating the camera often makes the artifact go away. Another "solution" is to use two cameras, one horizontal and one vertical and aggregating the results.

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@thanujadax Any other questions for this ticket? Looking forward to your update. Thanks!

thanujadax, 2019-05-23:
> If you haven't read it, I would advise you to take a look at the chapter 14 of [this [...]

Hi Antoine, thanks for your inputs. The parameter (DS Second Peak Threshold) suggested in the white paper does seem to have a positive effect, but it doesn't eliminate the problem, unfortunately. More than the holes, the bigger problem is that the 3D reconstruction shows parts of the mesh in random places in the image.

thanujadax, 2019-05-23:
> Hi @thanujadax
> Did you try loading High Accuracy JSON and increasing laser power to max?
> The preset can help invalidate wrong pixels while getting the projector pattern to be visible in the infrared [...]

Hi @dorodnic, thanks for your input!
The high accuracy setting unfortunately didn't solve the problem. But I'm going to try out IR pass through filters. Will be able to test it in one to two weeks
In theory, a tilt from the horizontal should reduce the problem to a certain extent, but for the grids I have to deal with, the effect of this tilt is not that useful. It could be that the repetition even at an angle is still difficult for the stereo matching algorithm to handle.

RealSenseCustomerSupport, 2019-06-14:
----------------------------------------------

@thanujadax Any good news for your testing with IR pass through filters? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

@thanujadax Any update?

RealSenseCustomerSupport, 2019-06-27:
----------------------------------------------

@thanujadax If no other questions for this ticket, close this accordingly. Thanks!

closed 2019-06-27 by RealSenseCustomerSupport