realsenseai/librealsense #3967: dynamic Colormap

ColinSmith123, 2019-05-13:
Hello everybody I have a smal problem. 

I need colored depth images that look like this : 

[image]
(dynamic black and white) 

the closest I can get is this:
[image]

is there a colormap preset for pyhton, that looks like the "dynamic black and white" in the Intel RS Viewer? 

Thanks :)

dorodnic, 2019-05-14:
Hi @ColinSmith123 
Yes, you can `set_option` on the colorizer object in python. I don't remember the exact options, but something like `rs2.option.color_scheme` and `rs2.option.histogram_equalization_enabled` (set to 1)

ColinSmith123, 2019-05-14:
Hello @dorodnic 

thanks for your fast answer. unfortunately, I guess I need a little more help. 

here is my code maybe you see what needs to be changed. 

import pyrealsense2 as rs
import numpy as np
import cv2

subject = '/A'
surface = '/C'
object = r'/003'
NrFrames = 27005

rgb = r'/rgb'
training = r'/training'
dirD = r'C:\DataSet\data';

pipeline = rs.pipeline()
config = rs.config()

#config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)   
#config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)   
config.enable_stream(rs.stream.color, 848, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)

depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

align_to = rs.stream.color
align = rs.align(align_to)

for i in range(NrFrames):
        frames = pipeline.wait_for_frames()
       
        aligned_frames = align.process(frames)       
        aligned_depth_frame = aligned_frames.get_depth_frame() 
        color_frame = aligned_frames.get_color_frame()
        
        if not aligned_depth_frame or not color_frame:
            continue
        
        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.1, beta=0.11), cv2.COLORMAP_HOT)
        images = np.hstack((color_image, depth_colormap))
        
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        key = cv2.waitKey(1)
        
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
        
        print(i)
        
        depthFlip = np.flipud(depth_colormap)
        rgbFlip = np.flipud(color_image)

        #cv2.imwrite(dirD + subject + surface + object + training + "/depth%d.jpg" % i, depth_colormap)
        #cv2.imwrite(dirD + subject + surface + object + rgb + "/rgb%d.jpg" % i, color_image)
        #cv2.imwrite(dirD + subject + surface + object + training + "/depthflip%d.jpg" % i, depthflip)
        #cv2.imwrite(dirD + subject + surface + object + rgb + "/rgbflip%d.jpg" % i, rgbFlip)
               
else:       
    cv2.destroyAllWindows()
    pipeline.stop()

Thanks a lot!!

ColinSmith123, 2019-05-15:
Meanwhile I found this in another issue ( How to use the function colorlizer in matlab or python? #2520 )

[image]

I tried to implement this and using the colorizer instead of cv-colormap works but I can't change the color_scheme. 

I get this error: 

  File "<ipython-input-28-5dcc2c65cec9>", line 58, in <module>
    colorizer.set_option(rs.option.color_scheme, 2);

TypeError: set_option(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.options, option: pyrealsense2.option, value: float) -> None

Invoked with: <pyrealsense2.colorizer object at 0x0000028EE3219DF8>, 1, 2

whats the problem? sorry I know I am not a good programmer and new to this field 

Any help would be appreciated :)

lramati, 2019-05-19:
What version of pyrealsense2 are you using?
In the meantime, so long as you don't want to change the color scheme mid-program, you should be able to just create the colorizer with the desired scheme using `colorizer = rs.colorizer(2)`

ColinSmith123, 2019-05-21:
@Iramati thanks aaaaallooott it works! 

Thanks to everybody else :)

closed 2019-07-22 by dorodnic