realsenseai/librealsense #3969: IMU and Depth Stream From .bag File ( PYTHON)

Bedrettin-Cetinkaya, 2019-05-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d435i } | 
| Firmware Version                   | (5.11.4) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<21>.<0> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
i try to get motion (accel,gyro) and depth data together by using following codes:

    def main():
    try:
        config = rs.config()
        rs.config.enable_device_from_file(config,test1.bag")
        pipeline = rs.pipeline()
        config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
        config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)
        config.enable_stream(rs.stream.gyro)
        config.enable_stream(rs.stream.accel)
        pipeline.start(config)
        '''
        align_to = rs.stream.depth
        align = rs.align(align_to)
        '''
        i = 0
        while True:
            print("Saving frame:", i)
            frames = pipeline.wait_for_frames()
            
            #aligned_frames = align.process(frames)
            
            depth_frame = frames.get_depth_frame()
            color_frame = frames.get_color_frame()
            
            depth_image = np.asanyarray(depth_frame.get_data(), dtype=float)
            depth_image = depth_image * 255 / np.amax(depth_image)

            color_image = np.asanyarray(color_frame.get_data(), dtype=float)

           
            gyro_frame = frames[0].as_motion_frame().get_motion_data()
            accel_frame = frames[1].as_motion_frame().get_motion_data()
            
            gyro = np.asanyarray([gyro_frame.x,gyro_frame.y,gyro_frame.z], dtype=float)
            print gyro
            accel = np.asanyarray([accel_frame.x,accel_frame.y,accel_frame.z], dtype=float)
            print accel            
            color_image = color_image[:,:,::-1]

            i += 1
            if i > 3:
                break
    finally:
        pass

    if __name__ == "__main__":
        main()`

it gives following error:

    "in main, gyro_frame = frames[0].as_motion_frame().get_motion_data()
     RuntimeError: null pointer passed for argument "frame_ref""

I could get motion and depth data separately, but i couldn't get them together. How can i achive get both motion and depth data same time with python?

Thanks.

Praneet1997, 2019-05-20:
I have also stuck with the same problem , not able to get both of them

RealSenseCustomerSupport, 2019-08-15:
----------------------------------------------

Hi @Praneet1997,

The similar issue with #4018. If no question, I will close this.

closed 2019-12-05 by RealSenseCustomerSupport