realsenseai/librealsense #3974: create default matcher loop crash

stephen-systemfriend, 2019-05-14:
SDK 2.20
Unity
Code using patch from PR #3972

when I run an object with the following code on it in unity in a pipeline with a point cloud attached to it, Unity crashes with no errors reported.
When I look at the log, I find 

` 14/05 16:02:52,852 DEBUG [10728] (device.cpp:47) Created default matcher
 14/05 16:02:52,852 ERROR [10728] (sync.cpp:204) Stream matcher not found! stream=Color
 14/05 16:02:52,852 DEBUG [10728] (sync.cpp:143) DISPATCH (TS: )--> Color 0 790.000000 
`
repeated ad nauseam, all with the same timestamp

Any idea what is causing this problem (it happens no matter what matcher I choose)

`
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using Intel.RealSense;
using System;

public class RSSoftwareDevTest : RsFrameProvider
{

    public float[] intrinsicsCoeffics = new float[] { 0, 0, 0, 0, 0 };

    //the device by which we generate our frames
    private SoftwareDevice framegenerator;
    private SoftwareSensor colorGenerator, depthGenerator;
    private VideoStreamProfile colorStream, depthStream;
    private Syncer sync;
    private int imgWidth, imgHeight, depthWidth, depthHeight;

    //for pipeline parameters
    private Context ctx;
    private Pipeline pipe;

    public int fps = 30;
    

    void Awake()
    {
        //initialise the log
        Intel.RealSense.Log.ToFile(Intel.RealSense.LogSeverity.Debug, @"debug.txt");
    }
    
    //this is a frame provider
    /// <summary>
    /// Notifies upon streaming start
    /// </summary>
    public override event Action<PipelineProfile> OnStart;

    /// <summary>
    /// Notifies when streaming has stopped
    /// </summary>
    public override event Action OnStop;

    /// <summary>
    /// Fired when a new frame is available
    /// </summary>
    public override event Action<Frame> OnNewSample;

    public Texture2D colorTex;
    public Texture2D depthTex;
    int frameNumber = 0;
    private ushort[] depthData;

    private bool initialising = true;

    // Update is called once per frame
    void Update()
    {
        if (!Streaming) return;
        if (initialising)
        {
            if (colorStream != null)
            {
                colorGenerator.Stop();
                colorGenerator.Close();
                colorStream.Dispose();
            }
            if (depthStream != null)
            {
                depthGenerator.Stop();
                depthGenerator.Close();
                depthStream.Dispose();
            }

            var intrinsics = new Intrinsics
            {
                width = colorTex.width,
                height = colorTex.height,
                ppx = imgWidth / 2,
                ppy = imgHeight / 2,
                fx = colorTex.width,
                fy = colorTex.height,
                model = Distortion.BrownConrady,
                coeffs = intrinsicsCoeffics
            };

            var colorProfile = new SoftwareVideoStream
            {
                type = Stream.Color,
                index = 0,
                uid = 101,
                width = colorTex.width,
                height = colorTex.height,
                fps = fps,
                bpp = 3,
                format = Format.Rgb8,
                intrinsics = intrinsics
            };

            colorStream = colorGenerator.AddVideoStream(colorProfile);

            var dpthIntrinsics = new Intrinsics
            {
                width = depthTex.width,
                height = depthTex.height,
                ppx = depthTex.width / 2,
                ppy = depthTex.height / 2,
                fx = depthTex.width,
                fy = depthTex.height,
                model = Distortion.BrownConrady,
                coeffs = intrinsicsCoeffics
            };

            var depthProfile = new SoftwareVideoStream
            {
                type = Stream.Depth,
                index = 0,
                uid = 100,
                width = depthTex.width,
                height = depthTex.height,
                fps = fps,
                bpp = 2,
                format = Format.Z16,
                intrinsics = dpthIntrinsics
            };

            depthStream = depthGenerator.AddVideoStream(depthProfile);
            depthGenerator.AddReadOnlyOption(Option.DepthUnits, 1.0f / 5000);
            
            depthGenerator.Open(depthStream);
            colorGenerator.Open(colorStream);
            sync = new Syncer();
            colorGenerator.Start(sync.SubmitFrame);
            depthGenerator.Start(sync.SubmitFrame);

            var extr = new Extrinsics();
            extr.rotation = new float[] { 1, 0, 0, 0, 1, 0, 0, 0, 1 };
            extr.translation = new float[] { 0, 0, 0 };

            depthStream.RegisterExtrinsicsTo(colorStream, extr);

            //load depth data once
            depthData = new ushort[depthTex.width * depthTex.height];
            var depthpix = depthTex.GetPixels32();
            for(int i =0; i < depthpix.Length; i++)
            {
                depthData[i] = (ushort)(depthpix[i].r*100);
            }
            initialising = false;

        }

        //update our position
        if (sync != null)
        {
            colorGenerator.AddVideoFrame<byte>(colorTex.GetRawTextureData(), colorTex.width * 3,
                3, System.DateTime.Now.Millisecond, TimestampDomain.HardwareClock, frameNumber, colorStream);
            depthGenerator.AddVideoFrame<ushort>(depthData, depthTex.width * 2,
               2, System.DateTime.Now.Millisecond, TimestampDomain.HardwareClock, frameNumber, depthStream);

            using (var frameset = sync.WaitForFrames())
            {

                var depthframe = frameset.FirstOrDefault(Stream.Depth);
                var colorframe = frameset.FirstOrDefault(Stream.Color);

                if (colorframe == null) Debug.LogWarning("colorframe was null");

                if (depthframe == null) Debug.LogWarning("depthframe was null");

                //send it using the RS pipeline
                var onNewSample = OnNewSample;
                if (onNewSample != null)
                {
                    if(depthframe != null && colorframe != null)
                        onNewSample(frameset.AsFrame());
                }
                else
                {
                    Debug.LogWarning("OnNewSample is null");
                }
            }
            frameNumber++;
        }
    }

    private void OnEnable()
    {
        ctx = new Context();
        imgWidth = colorTex.width;
        imgHeight = colorTex.height;
        depthWidth = depthTex.width;
        depthHeight = depthTex.height;

        framegenerator = new SoftwareDevice();
        depthGenerator = framegenerator.AddSensor("Depth");
        colorGenerator = framegenerator.AddSensor("Color");
        framegenerator.SetMatcher(Matchers.DLR_C);

        framegenerator.AddTo(ctx);
        
        pipe = new Pipeline(ctx);

        StartCoroutine(WaitAndStart());
    }

    IEnumerator WaitAndStart()
    {
       yield return new WaitForEndOfFrame();

        var cfg = new Config();
        cfg.EnableStream(Stream.Depth, depthWidth, depthHeight, Format.Z16, fps);
        cfg.EnableStream(Stream.Color, imgWidth, imgHeight, Format.Rgb8, fps);
        
        ActiveProfile = pipe.Start();

        Streaming = true;
        if (OnStart != null)
            OnStart(ActiveProfile);
    }

    void OnDisable()
    {
        OnNewSample = null;

        if (Streaming && OnStop != null)
            OnStop();

        if (ActiveProfile != null)
        {
            ActiveProfile.Dispose();
            ActiveProfile = null;
        }

        Streaming = false;
    }
    void OnDestroy()
    {
        // OnStart = null;
        OnStop = null;

        if (ActiveProfile != null)
        {
            ActiveProfile.Dispose();
            ActiveProfile = null;
        }
    }

}
`

aangerma, 2019-05-15:
Hello
Does the method Update() called, if not, you are not setting any video stream to the color and depth software sensor and this is explain why the syncer can't generate any matcher even though you called to SetMatcher().

stephen-systemfriend, 2019-05-15:
The order of Calls will be:
Awake()
OnEnable()
Update() //returns straight away because streaming isnt yet set
WaitAndStart()
Update()

I moved the call to SetMatcher to during the update, but it still was sad.

The crash happens when I call 
`colorGenerator.AddVideoFrame<byte>(colorTex.GetRawTextureData(), colorTex.width * 3,
                3, System.DateTime.Now.Millisecond, TimestampDomain.HardwareClock, frameNumber, colorStream);
`

aangerma, 2019-05-15:
if the Update() returns straight away so actually WaitAndStart() executed before update() which means you call to pipe.start() before you set the video streams to sensors,
and the syncer start to work before you initialized the sensors with their streams, you need to initialize the sensors as you do in update() and after it to start the stream.

stephen-systemfriend, 2019-05-16:
Ah ok, that stopped that crash. Now I've hit a different problem - 
Update now looks like
`
void Update()
    {
        if (colorStream == null)
        {
            var intrinsics = new Intrinsics
            {
                width = colorTex.width,
                height = colorTex.height,
                ppx = imgWidth / 2,
                ppy = imgHeight / 2,
                fx = colorTex.width,
                fy = colorTex.height,
                model = Distortion.BrownConrady,
                coeffs = intrinsicsCoeffics
            };

            var colorProfile = new SoftwareVideoStream
            {
                type = Stream.Color,
                index = 0,
                uid = 101,
                width = colorTex.width,
                height = colorTex.height,
                fps = fps,
                bpp = 3,
                format = Format.Rgb8,
                intrinsics = intrinsics
            };

            colorStream = colorGenerator.AddVideoStream(colorProfile);
            colorGenerator.Open(colorStream);
            colorGenerator.Start(sync.SubmitFrame);
        }
        if (depthStream == null)
        {
            var dpthIntrinsics = new Intrinsics
            {
                width = depthTex.width,
                height = depthTex.height,
                ppx = depthTex.width / 2,
                ppy = depthTex.height / 2,
                fx = depthTex.width,
                fy = depthTex.height,
                model = Distortion.BrownConrady,
                coeffs = intrinsicsCoeffics
            };

            var depthProfile = new SoftwareVideoStream
            {
                type = Stream.Depth,
                index = 0,
                uid = 100,
                width = depthTex.width,
                height = depthTex.height,
                fps = fps,
                bpp = 2,
                format = Format.Z16,
                intrinsics = dpthIntrinsics
            };

            depthStream = depthGenerator.AddVideoStream(depthProfile);
            depthGenerator.AddReadOnlyOption(Option.DepthUnits, 1.0f / 5000);

            var extr = new Extrinsics();
            extr.rotation = new float[] { 1, 0, 0, 0, 1, 0, 0, 0, 1 };
            extr.translation = new float[] { 0, 0, 0 };

            depthStream.RegisterExtrinsicsTo(colorStream, extr);
            depthGenerator.Open(depthStream);
            depthGenerator.Start(sync.SubmitFrame);
        }

        if (depthData == null)
        {
            //load depth data once
            depthData = new ushort[depthTex.width * depthTex.height];
            var depthpix = depthTex.GetPixels32();
            for (int i = 0; i < depthpix.Length; i++)
            {
                depthData[i] = (ushort)(depthpix[i].r * 100);
            }
        }

        if (!Streaming)
        {
            var cfg = new Config();
            cfg.EnableStream(Stream.Depth, colorTex.width, colorTex.height, Format.Z16, fps);
            cfg.EnableStream(Stream.Color, depthTex.width, depthTex.height, Format.Rgb8, fps);
            ActiveProfile = pipe.Start(cfg);

            Streaming = true;
            if (OnStart != null)
                OnStart(ActiveProfile);
        }
        
        //update our position
        if (sync != null)
        {
            colorGenerator.AddVideoFrame<byte>(colorTex.GetRawTextureData(), colorTex.width * 3,
                3, System.DateTime.Now.Millisecond, TimestampDomain.HardwareClock, frameNumber, colorStream);
            depthGenerator.AddVideoFrame<ushort>(depthData, depthTex.width * 2,
               2, System.DateTime.Now.Millisecond, TimestampDomain.HardwareClock, frameNumber, depthStream);

            using (var frameset = sync.WaitForFrames())
            {

                var depthframe = frameset.FirstOrDefault(Stream.Depth);
                var colorframe = frameset.FirstOrDefault(Stream.Color);

                if (colorframe == null) Debug.LogWarning("colorframe was null");

                if (depthframe == null) Debug.LogWarning("depthframe was null");

                //send it using the RS pipeline
                var onNewSample = OnNewSample;
                if (onNewSample != null)
                {
                    if(depthframe != null && colorframe != null)
                        onNewSample(frameset.AsFrame());
                }
                else
                {
                    Debug.LogWarning("OnNewSample is null");
                }
            }
            frameNumber++;
        }
    }
`
but it throws when I call pipe.Start(cfg) saying "Software device is streaming!"

if I instead call pipe.start(), then it instead throws during RSPointCloudRenderer.cs line 33 because the passed pipeline profile has no streams.
So then it runs, but will never produce the desired result (that is, a point cloud)

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

@stephen-systemfriend Sorry for the late response! Did you still get the issue of "Software device is streaming!"?

stephen-systemfriend, 2019-06-25:
Just upgraded to the latest version, still getting this error

stephen-systemfriend, 2019-06-25:
after digging around through the code for a while, I fixed it.
the correct code doesn't call 
        depthGenerator.Open(depthStream);
        depthGenerator.Start(sync.SubmitFrame);

or
        colorGenerator.Open(colorStream);
        colorGenerator.Start(sync.SubmitFrame);

because those are started by pipe.open(cfg).
instead what I had to do was call
pipe.open(cfg, sync.SubmitFrame) instead of pipe.open(cfg), and now it all works

closed 2019-06-25 by stephen-systemfriend