realsenseai/librealsense #3975: std::runtime_error with opencv

TaiBai0210, 2019-05-14:
I have an D435 camera, it can be used normally with the 2.17.1 sdk.
But when I want to combine it with CMU-OPENPOSE like this way
[image]

It came with std::runtime_error. Could you plz tell the problem about my code? thx

TaiBai0210, 2019-05-14:
here is the error 
[image]

dorodnic, 2019-05-14:
Hi @TaiBai0210 
Can you wrap the code in `try { } catch(std::runtime_error e){ std::cout << e.what(); } ` and share the exact error message?

TaiBai0210, 2019-05-21:
Thank you for helping me, I change my code try catch like you say 
[image]
here is the new error_msg
0x00007FFE9EEED7EB (opencv_world310.dll)处(位于 OpenPoseDemo.exe 中)引发的异常: 0xC0000005: 读取位置 0x0000000000000010 时发生访问冲突。
But I have already got the file like this
[image]
Could you please tell me how to deal with it? Thank you very much.

RealSenseCustomerSupport, 2019-06-02:
----------------------------------------------

Hi @TaiBai0210,

Please try to modify the code as follows. Any reason to input myVideoCapture to frame which is from "get435Frame()"?

  Mat colorMat(Size(colorFrames.as<rs2::video_frame>().get_width(), colorFrames.as<rs2::video_frame>().get_height()), CV_8UC3, (void*)colorFrames.get_data(), cv::Mat::AUTO_STEP);

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @TaiBai0210,

Any update? I will close this if no additional question.

closed 2019-12-05 by RealSenseCustomerSupport