realsenseai/librealsense #3978: D400 Series Firmware 5.11.4 and 5.11.6.200 Assign D435i Incorrect USB PIDs

oursland, 2019-05-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.11.4 and 5.11.6.200 | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  4.15.0-48-generic | 
| Platform                                 | PC |
| SDK Version                            |  N/A | 
| Language                            |  N/A      | 
| Segment			|  N/A     | 

Firmware versions 5.11.4 and 5.11.6.200 (possibly others) assign the D435i incorrect USB Product IDs.  In one instance it reports as a D430i and another it is the D400.  In both situations the devices fail to operate as expected with realsense-viewer, rs-fw-logger, and realsense-ros.

Reverting to firmware version 5.11.1.100 appears to correctly identify the device as a D435i.

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

Hi oursland,

Wonder what do you mean by not operated as expected...can you please help elaborate?

Also would you like to try the newly released RS2.22.0 with FW5.11.4 and see if it helps?
From what I have tested here, it works well for all the FW versions.

Thanks!

oursland, 2019-05-21:
The devices enumerate incorrectly causing librealsense to believe they have different specifications than the actual device.  Consequently I lost the IMU when it identified as the D400 and the RGB when it claimed to be the D430i.

I'm under a bit of a time crunch right now, so I'll have to provide lsusb output showing the incorrect PIDs at a later time.

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi oursland,

I am not aware of the issues, and not able to replicate it here (D435i w/ FW5.11.4, FW5.10.13). With both Ubuntu 16.04 and 18.04, no issue observed.

As you mentioned, this is off a PC with Ubuntu 18.04, right? Are you using the USB cable shipped with D435i or a different one.
Can you please help collect following info when it happens again:
a) take screenshots of RSViewer (if using it);
b) collect dmesg dump info, run rs-fw-logger and collect its dump, and then lsusb dump.

Thanks!

RealSenseCustomerSupport, 2019-06-06:
----------------------------------------------

@oursland Did you get through the issue? Could you please update? Thanks!

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@oursland Any update from your side?

RealSenseCustomerSupport, 2019-06-21:
----------------------------------------------

@oursland Any other questions for this ticket? Thanks!

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@oursland  No update from you for several weeks, will close the ticket at this point. Please feel free to create new ticket if you still have questions or issues. Thanks!

closed 2019-06-25 by RealSenseCustomerSupport