realsenseai/librealsense #3979: How to get IMU data for multi camera

AhnHyeongHyun, 2019-05-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                     | 
| Firmware Version                   | 05.11.04.00                                             | 
| Operating System & Version |   Win10                                                   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  PC                                                          |
| SDK Version                            |  { legacy / 2.21.0 }                          | 
| Language                               |  C++                                                      | 
| Segment			         |                                                             | 

### Issue Description
Hello.
I see the rs-examples 'motion' and 'multicam'
I try merge to 'motion' and 'multicam'  but faild.
I don't know what to do by using following codes:(examples/motion)
`
// Start streaming with the given configuration;

    // Note that since we only allow IMU streams, only single frames are produced
    auto profile = pipe.start(cfg, [&](rs2::frame frame)
    {
        // Cast the frame that arrived to motion frame
        auto motion = frame.as<rs2::motion_frame>();
        // If casting succeeded and the arrived frame is from gyro stream
        if (motion && motion.get_profile().stream_type() == RS2_STREAM_GYRO && motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F)
        {
            // Get the timestamp of the current frame
            double ts = motion.get_timestamp();
            // Get gyro measures
            rs2_vector gyro_data = motion.get_motion_data();
            // Call function that computes the angle of motion based on the retrieved measures
            algo.process_gyro(gyro_data, ts);
        }
        // If casting succeeded and the arrived frame is from accelerometer stream
        if (motion && motion.get_profile().stream_type() == RS2_STREAM_ACCEL && motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F)
        {
            // Get accelerometer measures
            rs2_vector accel_data = motion.get_motion_data();
            // Call function that computes the angle of motion based on the retrieved measures
            algo.process_accel(accel_data);
        }
    });
`
How to get IMU data for multi camera?
Sorry, My English is not good.

ev-mp, 2019-05-15:
@AhnHyeongHyun hello,
The `rs-multicam` demo does exactly this, and its flow is summarized as follows:
 - Create a `pipe` object for each camera connected.
 - Configure each pipe using the recommended setup (D435i: Depth+Color+Accel+Gyro)
 - Run the pipes object in loop -> collect and visualize all frames including motion.
Follow the example's code to see how the motion [data is retrieved](https://github.com/IntelRealSense/librealsense/blob/master/examples/example.hpp#L120)

`rs-multicam` and `rs-motion` use separate modes of operation - synchronous and asynchronous accordingly. Thus you first need to establish which is more suited for your needs.
Hence I may recommend a gradual approach:
- The synchronous calls for `pipe.poll_for_frames(..` employed in the `rs-multicam` sample is the straight-forward single-threaded way. So just modifying the demo to drop Depth and Color and leave IMU frames only would probably be a good starting point.
- Later on you can review the asynchronous way (using the callback you've posted) - it can provide certain advantages at the expense of a more complicated code.

AhnHyeongHyun, 2019-05-15:
@ev-mp Thank you reply.
I'll use your idea.

ev-mp, 2019-05-19:
@AhnHyeongHyun hi, do you have any follows up?

RealSenseCustomerSupport, 2019-08-15:
----------------------------------------------

Hi @AhnHyeongHyun,

any update?

closed 2019-11-01 by RealSenseCustomerSupport