realsenseai/librealsense #3980: version 2.21.0  imu（gyro/acce）timestamp is error

hitxuhao, 2019-05-15:
hardware:d435i 
software_version:2.21.0
problem:gyro/accel devices timestamp is error
motion example:
    auto profile = pipe.start(cfg, [&](rs2::frame frame)
    {
        // Cast the frame that arrived to motion frame
        auto motion = frame.as<rs2::motion_frame>();
        // If casting succeeded and the arrived frame is from gyro stream
        if (motion && motion.get_profile().stream_type() == RS2_STREAM_GYRO && motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F)
        {
            // Get the timestamp of the current frame
            double ts = motion.get_timestamp();
           // std::cout<<"gyro:\n"<<gyro_data<<std::endl;
            // Get gyro measures
            rs2_vector gyro_data = motion.get_motion_data();

            printf("gyro timestamp:%lf x:%f y:%f z:%f\n",ts,gyro_data.x,gyro_data.y,gyro_data.z);
            // Call function that computes the angle of motion based on the retrieved measures
            algo.process_gyro(gyro_data, ts);
        }
        // If casting succeeded and the arrived frame is from accelerometer stream
        if (motion && motion.get_profile().stream_type() == RS2_STREAM_ACCEL && motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F)
        {
            // Get accelerometer measures
            double ts = motion.get_timestamp();
            rs2_vector accel_data = motion.get_motion_data();
            printf("accel timestamp:%lf x:%f y:%f z:%f\n",ts,accel_data.x,accel_data.y,accel_data.z);
            //std::cout<<"acc:\n"<<accel_data<<std::endl;
            // Call function that computes the angle of motion based on the retrieved measures
            algo.process_accel(accel_data);
        }
    });

print result:
accel timestamp:184942.757784 x:1.274864 y:-9.767424 z:0.804145
gyro timestamp:1557827082567.210449 x:-0.006981 y:0.000000 z:0.005236
gyro timestamp:1557827082572.211670 x:-0.010472 y:0.000000 z:0.005236
gyro timestamp:1557827082577.211914 x:-0.006981 y:-0.001745 z:0.005236
gyro timestamp:1557827082582.209229 x:-0.012217 y:0.000000 z:0.003491
accel timestamp:184958.534785 x:1.274864 y:-9.806650 z:0.804145
gyro timestamp:1557827082587.209961 x:-0.005236 y:0.000000 z:0.005236
gyro timestamp:1557827082592.209961 x:-0.003491 y:0.000000 z:0.006981
gyro timestamp:1557827082597.190430 x:-0.006981 y:-0.001745 z:0.001745
accel timestamp:184974.312786 x:1.245445 y:-9.747809 z:0.764919
gyro timestamp:1557827082602.209473 x:0.000000 y:0.000000 z:0.000000
gyro timestamp:1557827082607.209229 x:0.000000 y:0.000000 z:0.000000
gyro timestamp:1557827082612.209961 x:0.001745 y:0.001745 z:0.000000
accel timestamp:184990.088787 x:1.255251 y:-9.747809 z:0.804145
gyro timestamp:1557827082617.208984 x:0.000000 y:0.005236 z:0.005236
gyro timestamp:1557827082622.208496 x:0.000000 y:0.001745 z:0.000000
gyro timestamp:1557827082627.208008 x:-0.005236 y:0.000000 z:0.001745
accel timestamp:185005.864787 x:1.294478 y:-9.767424 z:0.892405
gyro timestamp:1557827082632.225342 x:-0.008727 y:-0.005236 z:0.001745
gyro timestamp:1557827082637.206787 x:-0.013963 y:0.000000 z:0.000000
gyro timestamp:1557827082642.206055 x:-0.013963 y:0.000000 z:0.001745
accel timestamp:185021.641788 x:1.274864 y:-9.787037 z:0.853179
gyro timestamp:1557827082647.224854 x:-0.017453 y:-0.001745 z:0.010472
gyro timestamp:1557827082652.214600 x:-0.010472 y:-0.003491 z:0.005236
gyro timestamp:1557827082657.211426 x:-0.010472 y:0.003491 z:0.006981
accel timestamp:185037.417789 x:1.245445 y:-9.747809 z:0.745305
gyro timestamp:1557827082662.205078 x:-0.012217 y:-0.001745 z:0.003491
gyro timestamp:1557827082667.475830 x:-0.008727 y:0.000000 z:0.000000
gyro timestamp:1557827082672.456787 x:-0.005236 y:0.001745 z:0.001745
accel timestamp:185053.195790 x:1.225831 y:-9.767424 z:0.725692
gyro timestamp:1557827082677.456299 x:0.000000 y:0.005236 z:0.001745
gyro timestamp:1557827082682.456055 x:0.000000 y:0.000000 z:0.000000
gyro timestamp:1557827082687.454102 x:0.000000 y:0.000000 z:0.003491
gyro timestamp:1557827082692.456055 x:-0.000000 y:-0.005236 z:0.003491

absoulately:
timestamp is wrong, I'm disappointed about sdk

ev-mp, 2019-05-15:
@hitxuhao hello,
Given that you haven't filled in the issue template properly it is hard to address it with specificity,
but based on previous experience one viable hypothesis is that you're running a Linux distribution with an un-patched kernel version 4.8+. 
If that's the case then following the installation guide and using one of the two supported methods ([1](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages),[2](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites)) should align IMU timestamps by converting Gyro samples from Host to HW domain.

zjcs, 2019-05-17:
[image]
HI, @ev-mp , I am confused the same issue. 
I run the SDK on Win10 by install the latest realsense exe whose version is 2.21.0. 

It seems that different stream has different clock, is there some API to convert the stream's clock to system clock?

zjcs, 2019-05-17:
Sorry, the image uploading is failed. I will describe the D435iTimeStamp.png Info.

Camera Model: D435i
Firmware Version: 5.10.13.00
Operating System: Win10 
SDK Version: 2.21.0
Language: C++

Depth Stream TimeStamp: 1558094048043.5
Acc Stream TimeStamp    : 206.7
Gyro Stream TimeStamp  : 20675.4 

B.T.W, T265 Module works very well, every stream uses system clock.

ev-mp, 2019-05-19:
@zjcs hi,
The issue you're experiencing is due to a known issue that was resolved in FW version 5.11.2.
Please upgrade the firmware to the latest recommended FW and update.
Recommended FW 5.11.6.100 - [link](https://downloadcenter.intel.com/download/28573/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t)
Latest Firmware upgrade tools for Windows - [link](https://downloadcenter.intel.com/download/27514/Windows-Device-Firmware-Update-Tool-for-Intel-RealSense-D400-Product-Family)

@hitxuhao hello, did you manage to troubleshoot it ?

ev-mp, 2019-05-19:
@zjcs, regarding the other question:
Currently Librealsense delivers two separate timestamps for each D435i IMU frame (will change in upcoming v2.22):
- Host arrival timestamp - **always available** via metadata attribute RS2_FRAME_METADATA_TIME_OF_ARRIVAL
- Device/HW timestamp - available via `frame.get_timestamp()` API. This timestamp is provided *for free* on WinOs and requires a kernel patch on Linux. 
When HW timestamps cannot be retrieved Librealsense will automatically fallback to the Host timestamp, to make sure `frame.get_timestamp()` API remains persistent. In order to understand whether the acquired timestamp is host or hardware-originated call `frame.get_frame_timestamp_domain()` API.

zjcs, 2019-05-22:
@ev-mp  Thanks very much.
I update D435i Firmwire, but I can't find version 5.11.6.100 you recommened, so I try the lastest release version 5.11.6.200.

Now, (1) the timestamp of RGB and Depth are based ms, they are only host arrival timestamp. (2) the timestamp of Acc and Gyro are based ms, they can get HW timestamp and host arrival timestamp.

So, there still is some trouble of RGB and Depth timestamp. Do you meet the simillar issuse?

B.t.w, due to all frames host arrival timestamp are available now, I are able to work again, thank you again.

ev-mp, 2019-05-22:
@zjcs, I need to correct my previous statement - getting HW timestamps on Windows is `almost for free` - you need to run a script to enable it in PC's registry.
Check the installation guide for details - https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows

ev-mp, 2019-06-17:
@zjcs , this seem to be resolved, and I'm closing it.
Feel free to reopen as needed.

closed 2019-06-17 by ev-mp