realsenseai/librealsense #3985: hello-world program does not run successfully in relasense sdk 2.0

MattixLee, 2019-05-15:
hello,

In win10, realsense sdk 2.0 was installed,

rs-examples was opened in vs2013, when I compiled hello-world, nine error was occured.
[image]

I hope you can help me solve it. Thank you!

scizors, 2019-05-15:
did you build in vs2013 .
 
while building you had any error related to libusb

MattixLee, 2019-05-15:
@scizors 

hello,

yes, I used vs2013, I did not have  any error related to libusb.

Did you have the same error ?

scizors, 2019-05-15:
yes if I build librealsense2 in vs2015

Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB6006	"cmd.exe" exited with code 1.	libusb (ExternalProjectTargets\libusb\libusb)	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets	171

scizors, 2019-05-15:
Creating directories for 'libusb'
  Building Custom Rule C:/Users/rsr5le/Desktop/realsense/c++/librealsense-master/CMakeLists.txt
  CMake does not need to re-run because C:/Users/rsr5le/Desktop/realsense/librealsense-master/c++build/CMakeFiles/generate.stamp is up-to-date.
  Performing download step (git clone) for 'libusb'
  Cloning into 'libusb'...
  fatal: unable to access 'https://github.com/libusb/libusb.git/': Could not resolve host: github.com
  Cloning into 'libusb'...
  fatal: unable to access 'https://github.com/libusb/libusb.git/': Could not resolve host: github.com
  Cloning into 'libusb'...
  fatal: unable to access 'https://github.com/libusb/libusb.git/': Could not resolve host: github.com
  -- Had to git clone more than once:
  CMake Error at C:/Users/rsr5le/Desktop/realsense/librealsense-master/c++build/libusb-prefix/tmp/libusb-gitclone.cmake:66 (message):
            3 times.
    Failed to clone repository: 'https://github.com/libusb/libusb.git'
  
  
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code 1.

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

HI there,

For installing SDK and building samples, I used VS2015 and target to Windows 10 and have no issue of running all the samples after installed RS2.19.1 SDK. Also I am using VS2015.

a) Can you please share the info of which RS SDK release you used?
b) Any change you made after the installation? For the project's configuration settings, are they set up correctly to the installation path info?
c) Do you have another machine to try VS2015?

Thanks!

MattixLee, 2019-05-21:
@RealSenseCustomerSupport 
Hi, 

When I used vs2019 to compile realsense SDK, no errors occured.

Thanks!

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi MattixLee,

Thanks for the update. Glad that it works for you now.
Closing this one.

Thanks!

closed 2019-05-23 by MattixLee