realsenseai/librealsense #3986: how to get the intel D435 intrinsic and extrinsic calibration information in python

scizors, 2019-05-15:
CAMERA D435
OS windows
Python 3.6
pyrealsense2

I have python wrappers installed.
I have RGB Camera and stereo camera Left and right
I need the intrinsic and extrinsic  parameters for all the 3 camera 

I use 
profile = pipeline.get_active_profile()
print(profile) 
depth_profile = rs.video_stream_profile(profile.get_stream(rs.stream.depth))
print(depth_profile) 
depth_intrinsics = depth_profile.get_intrinsics()
print(depth_intrinsics)   
w, h = depth_intrinsics.width, depth_intrinsics.height
print(w,h)   

Output
<bound method PyCapsule.enable_stream of <pyrealsense2.pyrealsense2.config object at 0x0000029BB1C6CA40>>
<bound method PyCapsule.enable_stream of <pyrealsense2.pyrealsense2.config object at 0x0000029BB1C6CA40>>
<pyrealsense2.pyrealsense2.config object at 0x0000029BB1C6CA40>
<pyrealsense2.pyrealsense2.pipeline_profile object at 0x0000029BB1C6CAB0>
<pyrealsense2.video_stream_profile: Depth(0) 640x480 @ 30fps Z16>
width: 640, height: 480, ppx: 321.635, ppy: 241.618, fx: 385.134, fy: 385.134, model: Brown Conrady, coeffs: [0, 0, 0, 0, 0]
640 480
<pyrealsense2.pyrealsense2.colorizer object at 0x0000029BB1CD4458>

scizors, 2019-05-15:
I also did the same for color output 
  Get stream profile color and camera intrinsics
<pyrealsense2.pyrealsense2.pipeline_profile object at 0x0000023732FF3730>
<pyrealsense2.video_stream_profile: Color(0) 640x480 @ 30fps BGR8>
width: 640, height: 480, ppx: 320.503, ppy: 237.288, fx: 611.528, fy: 611.503, model: Brown Conrady, coeffs: [0, 0, 0, 0, 0]
640 480
<pyrealsense2.pyrealsense2.colorizer object at 0x0000023732FF35E0>

scizors, 2019-05-15:
I would like to know there is this correct procedure for finding the internsic values 

 How can I do it for the stereo camera Left and Right seperately

dorodnic, 2019-05-15:
Hi @scizors 
From pipeline profile you can get everything you enabled (via config). 
You can also always get the device object, enumerate over sensors and stream profiles and get intrinsics for each supported mode.
You can also get extrinsics between any two stream profile objects by using get_extrinsics_to

closed 2019-05-16 by scizors