realsenseai/librealsense #3987: T265 manual exposure control

mikonse, 2019-05-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5502 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |   4.15.0  | 
| SDK Version                            |  2.21.0 | 
| Language                            |  C++ | 

### Issue Description
Basically the title. How does one manually control the auto exposure / exposure time of the T265 fisheye cameras through the C++ or C API. As far as I've found out it is not possible via the standard librealsense API since the approach as described in the [how to]( https://github.com/IntelRealSense/librealsense/blob/2f6a26aa784cc140f12b9144fdc2fb2171d410fa/examples/sensor-control/api_how_to.h#L147-L184) does not seem to be supported on the T265. All sensor functions show up as not supported.

MartyG-RealSense, 2019-05-15:
@dorodnic said on Jan 28 2019 that "T265 performs auto-exposure on board.  Actual exposure and gain are reported with every frame via metadata.  If you are implementing your own 6-DOF algorithm (and not enabling the POSE stream) you can also set manual exposure value (this is not yet in LRS API, but should be possible according to device specifications)"

I would guess that LRS API is shorthand for Librealsense API. :)

https://github.com/IntelRealSense/librealsense/issues/3129#issuecomment-458203353

dorodnic, 2019-05-15:
Hi @mikonse 
Well, yes, the above is true. Our SLAM algorithm is fine-tuned with the built-in auto-exposure, so it will refuse to work with anything else (@bfulkers-i maybe can elaborate on why)
If you are running your own SLAM on top of fisheye and IMU data, this should be possible, but I need to check that it is exposed, since its a corner case scenario.

mikonse, 2019-05-15:
As you mentioned, running SLAM on top of the visual and imu data that is provided by the t265 is exactly my use case. 

Regarding access to low level API functions on the T265 I could only find a reference to the [device drivers](https://github.com/IntelRealSense/librealsense/blob/master/third-party/libtm/libtm/include/TrackingDevice.h#L508-L515) in the documentation. It however only points to the source code of the driver itself and I'm not quite sure how to access that when using the default librealsense installation. Is there a build flag to also expose the third party header files (which is where the driver code is located) when building manually?

dorodnic, 2019-05-16:
Please pull #3992 to unlock this functionality. We don't recommend using the driver directly, librealsense wraps it in a more convenient package. I need to clean-up the implementation a bit, but it should be functional for your needs.

mikonse, 2019-05-22:
Thank you very much, your patch is working perfectly. Is it going to be merged into the main branch eventually or will it stay a separate patch?

RealSenseCustomerSupport, 2019-06-19:
----------------------------------------------

Hi @mikonse,

Please have a try of the latest librealsense. The control option should be there.

closed 2019-06-19 by RealSenseCustomerSupport