realsenseai/librealsense #3988: pipe.start() leads to RuntimeError: Couldn't resolve request

RoWe90, 2019-05-15:
------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   | 05.11.01.100                                           | 
| Operating System & Version |   Win 10                                                  | 
| Platform                                 | PC                                                           |
| SDK Version                            |  2.16.1.251                                             | 
| Language                                |  python                                                  | 
| Segment			          |  others                                                   | 

### Issue Description
As the title says, when starting my script i get the above mentioned error. 
I made sure that the camera is connected via USB 3 (more precisely USB 3.2).
I also have the same problem with a D435.
The problem also seems to be getting worse, in the beginning i only had that issue when running my script from PyCharm, but was able to get around that by using IDLE to start it. But now, even that isn't working anymore.

Here is a snippet of the code for initializing the camera:

```
import os
import math
import cv2  # state of the art computer vision algorithms library
import numpy as np  # fundamental package for scientific computing
import pyrealsense2 as rs  # Intel RealSense cross-platform open-source API
import image_manipulation_functions as im  # custom functions for image manipulation
import time
import transformations as tf

print("Environment Ready")

# Legen einer 'Pipeline' zur Kamera
pipe = rs.pipeline()
# Konfigurieren der Kamera
cfg = rs.config()  # Anlegen einer config-Variablen um die pipeline fuer die Farb- und Tiefenkamera zur bearbeiten.

# Start streaming

cfg.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
cfg.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
#cfg.enable_device_from_file(os.path.join(__location__, Aufnahmen[4]))
cfg.enable_stream(rs.stream.pose)
# ----------------------------------------------------------------------
# Setup -Routine
# ----------------------------------------------------------------------
global profile
profile = pipe.start(cfg)

# Skip 5 first frames to give the Auto-Exposure time to adjust
for x in range(50):
    pipe.wait_for_frames()
    print("Test")

# Umwandlung des Tiefenskala in Array
# ----------------------------------------------------------------------------
depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
# -----------------------------------------------------------------------------
#more code...
```
I hope someone can help me out here :) thanks in advance!

dorodnic, 2019-05-15:
Hi @RoWe90 
At the moment `pose` stream is only provided by the T265 device. 
You can get the IMU streams from D435i.

closed 2019-06-03 by dorodnic

dwiel, 2019-10-17:
It would be nice if the error message was more helpful, such as "Pose estimation not available from your device D435i". I had to google this obscure error message for a while to figure out what was going on.