realsenseai/librealsense #3990: Dynamic calibration not working

riteshkhrn, 2019-05-16:
Hi,

Fix for this!

OS - 16.04

/usr/bin/Intel.Realsense.DynamicCalibrator -cli

Serial Number: 817412071622
Device Name: Intel RealSense D435
Device PID:  0B07
FW version:  05.10.06.00

Calibration Type: Targeted
Laser Power: On
Auto Exposure: On
AE SetPoint: Auto
Resolution: 1280x720x30

Error: calibration has been cancelled due to timeout! 
If you prefer, try longer timeout in seconds using the -o option on the command line or the advanced settings in GUI.
It's likely the device is severely shifted and hard for dynamic calibrator to recover. 
Please try latest technician calibration tool.
Segmentation fault (core dumped)

RealSenseCustomerSupport, 2019-05-28:
----------------------------------------------

Hi @riteshkhrn,

Since the dynamic calibration needs a target chart, please find the pdf in /usr/share/doc/librscalibrationtool/target/ (or, in below link) and print it. Or, you can display the target chart in your phone for the calibration.
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/print-target-fixed-width.pdf

RealSenseCustomerSupport, 2019-05-28:
----------------------------------------------

Refer to below guide with more detail.
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_User_Guide.pdf

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

Hi @riteshkhrn,

Is there any additional question or issue you use the dynamic calibration? If not, will close this.

riteshkhrn, 2019-09-20:
I tried with the chart. Still not working

RealSenseCustomerSupport, 2019-09-25:
----------------------------------------------

Hi @riteshkhrn,

1. Is dynamic calibration v2.6.8.0 used?
2. To run "Intel.Realsense.DynamicCalibrator.exe" with/without "-cli" option both is failed? Did you also try to increase the test time with the option "-t"?
3. Please give the logs in "~\DyCalibResult"
If dynamic calibration is still failed, probably, the twist/bending of your camera is very serious. Then, the factory calibration is required.

RealSenseCustomerSupport, 2019-09-25:
----------------------------------------------

You can also refer to below with troubleshooting.
https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide

ramilmsh, 2019-11-07:
> 3\. Please give the logs in "~\DyCalibResult"

```
Serial Number: <serial number>
Device Name: Intel RealSense D435
Device PID:  0B07
FW version:  05.11.15.00

Calibration Type: Targeted
Laser Power: On
Auto Exposure: On
AE SetPoint: Auto
Resolution: 1280x720x30
```

I have a similar problem. My calibration is stuck after printing Resolution: 1280x720x30 or (\#13 Resolution: 1280x720x30 in GUI). It is just a black screen with no feed. I have firmware v05.11.15.00, librealsense v2.29.0, and dynamic calibration tool v2.6.8.0. There is no camera feed or anything on the calibration screen

[image]

(valid serial number is present, simply photoshopped out)

The system manager shows almost no load on the CPU, yet GUI becomes laggy as soon as the script is running. It eventually times out and prints (in the terminal and in logs) message from the original post.

realsense-viewer works just fine (so does ros interface), there is no visible physical damage and, other than wobbly flat surfaces, point cloud seems ok too.

ramilmsh, 2019-11-08:
@RealSenseCustomerSupport , the question is withdrawn. I have searched the issues some more and it turned out it was USB 2.1 issue, as soon as I found a USB 3 cable, calibration tool worked just fine.

But, on that note, could I make a request for some sort of descriptive error message? like "you are using USB 2.1, realsense might not work properly"?

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

Hi @ramilmsh,

Thanks for the suggestion. Will also evaluate if it's necessary to support DC for USB2.

mamoru1126, 2019-12-06:
Hi @RealSenseCustomerSupport  
please help me, dynamic calibration does not work.

OS ubuntu 16.04
dynamic calibration ver 2.6.8.0
device d435i
using USB 3.0 cable

After press start calibration
output
#1 Serial number: <Serial number>
#2 Device name: Intel RealSense d435I
#3 Device PID: 0B3A
#4 FW version: 05.11.06.250
255.255.255.255
#5 Caribration Type; Targeted
#6 Laser Power: Off
#7 RGB: Yes
#8 Auto Exposure: On
#9: IR setpoint: 1000. (min: 0, max: 4095, default: 1000)
#10: RGB brightness: O, (min: -64, max: 64, default: 0)
#11:  AE SetPoint: Auto
#12: Failed to start caribration

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

Hi @mamoru1126,

1. Is the camera recognized as a USB3 device? Please also run "rs-enumerate-devices" to show the result.
2. Did you have a target chart printed or showed on mobile phone?
Refer to below link with more info.
https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide

mamoru1126, 2019-12-06:
@RealSenseCustomerSupport 
Thanks your reply

> 1. Is the camera recognized as a USB3 device? Please also run "rs-enumerate-devices" to show the result.

$ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	843112073535
    Firmware Version              : 	05.11.06.250
5.11.6.250
    Recommended Firmware Version  : 	05.11.06.250
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	852323053309

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 2	  1280x800	@ 30Hz	   Y8
    Infrared 1	  1280x800	@ 30Hz	   Y8
    Infrared 2	  1280x800	@ 25Hz	   Y16
    Infrared 1	  1280x800	@ 25Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y8
    Infrared 2	  1280x800	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 2	  1280x720	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 6Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 2	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared 2	  848x480	@ 15Hz	   Y8
    Infrared 2	  848x480	@ 6Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 2	  848x100	@ 100Hz	   Y8
    Infrared 1	  848x100	@ 100Hz	   Y8
    Infrared 1	  640x480	@ 90Hz	   Y8
    Infrared 2	  640x480	@ 90Hz	   Y8
    Infrared 2	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 2	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 2	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 2	  640x480	@ 6Hz	   Y8
    Infrared 2	  640x400	@ 25Hz	   Y16
    Infrared 1	  640x400	@ 25Hz	   Y16
    Infrared 1	  640x400	@ 15Hz	   Y16
    Infrared 2	  640x400	@ 15Hz	   Y16
    Infrared 2	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 60Hz	   Y8
    Infrared 2	  640x360	@ 60Hz	   Y8
    Infrared 1	  640x360	@ 30Hz	   Y8
    Infrared 2	  640x360	@ 30Hz	   Y8
    Infrared 1	  640x360	@ 15Hz	   Y8
[... 233 lines cut ...]
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 63Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 250Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F

I have a target chart showed on mobile phone.

OleguerCanal, 2019-12-07:
I am having the same problem as @mamoru1126 on ubuntu 18.04, dynamic calibration ver 2.6.8.0,
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	843112071657
    Firmware Version              : 	05.11.15.00
    Recommended Firmware Version  : 	05.11.15.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	852323053137
    Firmware Update Id            : 	852323053137

MarkWeav44, 2019-12-07:
Same issue here as well! I'm using a D435 and Dynamic Calibrator v2.6.8.0...
Confirm connected to USB 3 and have the Calibration target printed on a flat hard surface.
I just ordered a second camera that will be delivered on Monday and will return this one if that fixes the issue. In the meantime, I would be very interested if anyone has found a fix!

RealSenseCustomerSupport, 2019-12-11:
----------------------------------------------

HI @OleguerCanal, @MarkWeav44,

Do you see the images captured in the GUI when you move the camera or target chart?

OleguerCanal, 2019-12-12:
I can see the RGB, Depth and Gyro information without problems when I run realsense-viewer or subscribe to the realsense-ros topics. But when I run `/usr/bin/Intel.Realsense.DynamicCalibrator`
And click on `Start Calibration` I get the Failed to start Calibration Error as seen:
[image]

Thank you a lot

ninamwa, 2020-02-10:
Hi. Did anyone find a solution for this? I am having the exact same problem using my D435 camera. 
In the GUI it says "Failed to start calibration", but if using the CLI version I get the message: 

Serial Number: 831612070671
Device Name: Intel RealSense D435
Device PID:  0B07
FW version:  05.12.02.100

Calibration Type: Targeted
Laser Power: On
Auto Exposure: On
AE SetPoint: Auto
Resolution: 1280x720x30

A black screen appears, and nothing more happens.

ramilmsh, 2020-02-11:
> Hi. Did anyone find a solution for this? I am having the exact same problem using my D435 camera.
> In the GUI it says "Failed to start calibration", but if using the CLI version I get the message:
> 
> Serial Number:
> Device Name: Intel RealSense D435
> Device PID: 0B07
> FW version: 05.12.02.100
> 
> Calibration Type: Targeted
> Laser Power: On
> Auto Exposure: On
> AE SetPoint: Auto
> Resolution: 1280x720x30
> 
> A black screen appears, and nothing more happens.

make sure you are using a USB 3 cable (it will have blue plastic inside)

ninamwa, 2020-02-11:
> make sure you are using a USB 3 cable (it will have blue plastic inside)

Both the USB cable and port is blue inside (=USB3) so I think the problem must be something else..

VivienDelmon, 2020-02-14:
I also have issue to calibrate my D435 cameras. By using the GUI tool on Windows it freeze after scale calibration.
When I use the Intel.Realsense.DynamicCalibratorCLI.exe it also fails before RGB calibration but a CalibrationTable.xml file is created under DyCalibResult/<sn>/DC1/ with values that differ from those of the camera. I upload them to the camera using Intel.Realsense.CustomRW.exe -w -f CalibrationTable.xml and it seems to work. But this workflow is not really good ...

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

(1) There is still chance that D435 enumerated as USB2 device when plug into USB3 port if there signal integrity issue during enumeration process.
so before doing dynamic calibration,  you can run librealsense viewer to check if D435 is enumerate as either USB3 or USB 2 device
the DC 2.6.8 only support DC when D435 is enumerated as USB3 device

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

@VivienDelmon
Could you provide the calibration table before and after your dynamic calibration?
to restore to previous calibration table (factory setting), you can use
Intel.Realsense.CustomRW.exe -g
before reset to factory setting..
you can use
Intel.Realsense.CustomRW.exe -r -f afterDC.xml
to save the table after DC first

by the way, what is the reason you perform the dynamic calibration?
ex : you encounter some issue related to depth image quality?

do you have another Realsense camera that can perform the DC to see if this is device dependent?

ninamwa, 2020-02-25:
> Hi. Did anyone find a solution for this? I am having the exact same problem using my D435 camera.
> In the GUI it says "Failed to start calibration", but if using the CLI version I get the message: [...]

I was using the calibration tool on Ubuntu 18, following the "Installation on Linux" guide in the manual. When switching to Windows 8, and installing the software there, everything worked perfectly. A friend of mine also tried using the calibration tool in Ubuntu 18, which did not work (only able to read the calibration parameters 1 of 5 times, and never able to save the parameters after calibrating). Switching to Windows solved her problem as well. Might be something to test for other who are struggelig :)

hwb0314, 2020-03-03:
> I can see the RGB, Depth and Gyro information without problems when I run realsense-viewer or subscribe to the realsense-ros topics. But when I run `/usr/bin/Intel.Realsense.DynamicCalibrator`
> And [...]

@RealSenseCustomerSupport 
I need help !!!  Did anyone find a solution for this? I have the exact same problem as @OleguerCanal  when using my D435i device.

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Please use viewer to check
-if the device is enumerated as USB3 or USB 2 device
by checking upper left following the device name
ex Intel Realsense D435  3.2 means at USB 3 mode

the dynamic calibration only support USB3 mode
if you see device is at USB2 mode
ex Intel Realsense D435 2.x

you can run hardware Reset you can find it under "More"table at the viewer's upper left
then see if the device can re-enumerate as USB3 device

-if device able to streaming the depth and RGB without issue using viewer

stbjelcevic, 2020-03-09:
I am having the same issues. I have 4 cameras - 3 d435i's and 1 d435 and the dynamic calibration tool doesn't work for any of them. 

On Ubuntu 18.04.

> Please use viewer to check
> -if the device is enumerated as USB3 or USB 2 device
> by checking upper left following the device name
> ex Intel Realsense D435 3.2 means at USB 3 mode
>
> the dynamic [...]

I have checked on all of these points and each camera is able to stream and is enumerated as USB 3.x. I only have 1 camera plugged in at a time when I try to calibrate.

Perhaps I should see if I can get this working on windows?

RealSenseCustomerSupport, 2020-04-10:
----------------------------------------------

Hi @stbjelcevic
Thanks for your feedback
We are able to locate the root cause the main root cause is the DC 2.6.8 is not support kernel 5.3 yet.
We will release newer version of DC in the near future.
before that you can use windows 10 or Ubuntu 16.04 with older version of kernel for DC

closed 2020-05-04 by RealSenseCustomerSupport