realsenseai/librealsense #3991: Acceleration Range Issue on UAV

xuhao1, 2019-05-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i  | 
| Firmware Version                   |2.21.0| 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  Robot/UAV                       | 

### Issue Description

Dear Realsense Developer,
I am from HKUST UAV Group, and we are working on use RealSense D435i with our open-source visual inertial odometry [VINS-Fusion](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion) and fly it on UAV. 
In most case, like moving on a small car or hold by hand, the D435i works well and robust with our VINS. 
But when we use D435i on UAV, we found that the acceleration limit IMU of D435i is set to -4g to 4g, which is too small for UAV especially for landing and takeoff.
When we takeoff rapidly or landing, the IMU will saturation and then the VINs will fail.

For applying D435i on UAV, bigger IMU  acceleration range is very important. Is that possible for setting the acceleration range bigger?

MartyG-RealSense, 2019-05-16:
I did some research on the subject of acceleration range.  The Bosch manufacturer data sheet for the Bosch BMI055 IMU used in the D435i and T265 says that the component supports up to +- 16g, selectable via a digital interface.  So the +- 4g range may have been set by the RealSense engineering team - maybe @dorodnic can comment on this.

Apparently, the ideal situation is to have the range as low as possible.  The lower the range, the more sensitive the measurement is.  If the range is set high then it can cause problems, such as the vehicle moving faster than the sensor can keep up with (e.g the vehicle reaches the ground before the sensor detects the arrival).

xuhao1, 2019-05-18:
Actually we trying to make the vehicle touch  ground slower, but it's impossible to control it always, and it's not so robust and safe
Is that possible to provide a setable accelaration range option?

xuhao1, 2019-05-18:
There's balance between range and the sensitive, and for our application, we want more range than 4g. In our previous experience, bigger range and lowwer sensitive is enough for UAV

xuhao1, 2019-05-26:
otherwise, is that possible to provide more camera output frequency than 30hz?

dorodnic, 2019-05-27:
Hi @xuhao1 
We will look into making range configurable, but it’s unlikely to be implemented very soon. 
Both RGB and depth cameras can go beyond 30 FPS, but you might need to limit the resolution a bit 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

xuhao1, 2019-05-28:
Thank for you reply. Actually we want 20hz.

MartyG-RealSense, 2022-04-23:
Hi @xuhao1  I apologize that you did not receive further communication about this 2019 case.  It will be closed after 7 days from the time of writing this if we do not receive further comments.  Thanks!

closed 2023-07-18 by MartyG-RealSense