realsenseai/librealsense #3995: rs-latency-tool  :  out of range value for argument "index" when using USB CAM

patrickpoirier51, 2019-05-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {LOGITECH C920 } | 
| Firmware Version                   | N/A| 
| Operating System & Version |  UBUNTU 18 | 
| Kernel Version (Linux Only)    |Linux nvidia-nano 4.9.140-tegra | 
| Platform                                 |  NVIDIA Jetson NANO |
| SDK Version                            |  { 2.2.3 } | 
| Language                            |  {C}  | 
| Segment			| N/A | 
### Issue Description
<Describe your issue / question / feature request / etc..>
rs-latency-tool  :  out of range value for argument "index" when using USB CAM (Logitech C920)
: HD Pro Webcam C920 as /devices/70090000.xusb/usb1/1-2/1-2.1/1-2.1:1.0/input/input3
ls /dev/vid*
/dev/video0

**Error Message:**
rs-latency-tool 
RealSense error calling rs2_create_device(info_list:0x557111e5f0, index:0):
    out of range value for argument "index"

**Modified  rs-latency-tool.cpp according to wiki**
    // Start RealSense camera
    // Uncomment the configuration you wish to test
    //pipeline pipe;
    //config cfg;
    //cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_BGR8);
    //cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
    //cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 30);
    //cfg.enable_stream(RS2_STREAM_INFRARED, 640, 480, RS2_FORMAT_Y8);
    //cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_Y8);
    //pipe.start(cfg);

    // To test with a regular webcam, comment-out the previous block
    // And uncomment the following code:
    context ctx;
    auto dev = ctx.query_devices().front();
    auto sensor = dev.query_sensors().front();
    auto sps = sensor.get_stream_profiles();
    stream_profile selected;
    for (auto& sp : sps)
    {
        if (auto vid = sp.as<video_stream_profile>())
        {
            if (vid.format() == RS2_FORMAT_BGR8 &&
                vid.width() == 640 && vid.height() == 480 &&
                vid.fps() == 30)
                selected = vid;
        }
    }
    sensor.open(selected);
    syncer pipe;
    sensor.start(pipe);

dorodnic, 2019-05-16:
I think by default `query_devices` does not return non-RealSense devices, but there is a flag you can pass it to get these.

patrickpoirier51, 2019-05-16:
Thanks for response, and how do we pass these flags ? 
Within API code or as an argument ?

Regards

closed 2019-05-28 by RealSenseCustomerSupport