realsenseai/librealsense #3997: The pattern question for real sense camera

zhoushuai123, 2019-05-16:
Dear Sir/Madam,

Currently, I have a project regarding the simulating of the Real Sense D415 in Unreal Engine.

The pattern influences the accuracy of the stereo camera image output and reconstruction output.

Could Real Sense open&provide its pattern for usage?  

Thanks for any suggestion&information:-)

Best regards,
Mingchuan

MartyG-RealSense, 2019-05-16:
If you are asking for the camera's algorithms, these are not likely to be made available, as algorithms tend to be kept private by Intel as part of their intellectual property (IP) and not made open-source.

However, the RealSense SDK does provide a means via its **software_device** sample program of simulating camera hardware in software.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

zhoushuai123, 2019-05-16:
Thanks for your quick answer, I will check the link. However, I am not asking for the algorithm. I would like to ask whether I can get the pattern of the projector:-) . I saw in the image from infrared camera that it is kind of repetitive pattern with dots.

MartyG-RealSense, 2019-05-16:
Thank you very much for your clarification.  The dot pattern provided by the projector is a semi-random pattern of dots.  

Intel have an excellent illustrated camera tuning guide which contains recommendations about the dot pattern.  This information is provided in an extract below.

https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf?language=en_US

[image]

**PROJECTION PATTERNS**

[image]

[image]

dorodnic, 2019-05-16:
I’m not sure the pattern is stored anywhere in the system, but you can get a sense of it by taking snapshot of blank wall in ir. Unlike classic structured light, the pattern in theory should not matter, in the sense that the camera does not need to know what is the pattern. The D43x for instance have a different type of pattern from the d41x.
Of course different patterns do cause slight differences in behavior but so do different scenes. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2019-06-14 by RealSenseCustomerSupport