realsenseai/librealsense #3999: Missing Frame

Bedrettin-Cetinkaya, 2019-05-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400                                                      | 
| Firmware Version                   | 5.11.4                                                      | 
| Operating System & Version |   Win (/10)                                               | 
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                 | PC                                                           |
| SDK Version                            |  legacy / 2.21.0                         | 
| Language                            |  python                         | 
| Segment			|  others                      | 

I recorded .bag file ( depth, 2 infrared, color and 2 motion data ). When i printed frame timestamp via .py script, each run shows different data.  Data examples are as follows:

First run:
    `768786.641515
    768819.979517
    768853.317519
    768886.65452
    768953.330523
    768986.668525
    769020.0055259999
     769053.343528
     769086.68153
     769120.018531`

Second run:
    `768753.303514
    768786.641515
    768819.979517
    768853.317519
    768886.65452
    768919.992522
    768953.330523
    769020.0055259999
`

For motion data, missing frames are increased.

my .py scripts:

      config = rs.config()
      rs.config.enable_device_from_file(config, test_1.bag")
       pipeline = rs.pipeline()
       config.enable_stream(rs.stream.gyro)
       config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
       config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30) 
       config.enable_stream(rs.stream.accel)
       profile = pipeline.start(config)

       while True:
           frames = pipeline.wait_for_frames()        
           print(frames[0].get_timestamp())

what is the reason of this missing frame? How can i prevent it?

dorodnic, 2019-05-16:
This is significant amount of data, the system is most likely not fast enough to handle it. 
Solutions might include recording to an SSD or a RAM-Drive, making sure you use release version of the code, and monitoring general CPU consumption during recording.

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@Bedrettin-Cetinkaya Any update from your side regarding to Dorodnic's reply? Looking forward to your feedback. Thanks!

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@Bedrettin-Cetinkaya Do you still need support on this? Could you please update? Thanks!

Bedrettin-Cetinkaya, 2019-06-17:
Thanks, i solved the problem.

closed 2019-06-17 by Bedrettin-Cetinkaya