realsenseai/librealsense #4001: D435i not Recongized Properly

kat-mcg-lin, 2019-05-16:
Currently my D435i is only being recognized as a USB2.1 device and therefore I can't use some of the functions properly - i.e. the rs-data-collect 

I found this below about how it has to be recognized as a USB3 to work- how to I make it be recognized properly?? (I am working with a jetson nano dev kit)

According to the log you published, the device was connected as a USB 2 and it is behavior in this case is not well defined.
Assuming you connect it to a USB 3 jack, you should try to reconnect again and check how it was connected using linux tools, like this:
```
lsusb -d 8086: -v | grep -i bcdUSB
```
If it shows:
`bcdUSB               2.10` That's not good.
`bcdUSB               3.20` is good.

If you can run realsense-viewer, it also shows if it was connected as a usb2.1 or 3.2

_Originally posted by @doronhi in https://github.com/IntelRealSense/realsense-ros/issues/635#issuecomment-467754466_

MartyG-RealSense, 2019-05-16:
If the camera is in a USB 3 port but is being detected as USB 2, please first try inserting the camera into the port quickly and firmly instead of slowly.  This should increase the chances of it being correctly identified as USB 3.

kat-mcg-lin, 2019-05-17:
I just tried and it is still recognized only as "Intel Realsense USB2" with USB Descriptor of 2.1

MartyG-RealSense, 2019-05-17:
Are you using the official short-length cable supplied with the camera please?  Some users have found that they could only get USB 2 when using their own choice of cable, and solved the problem by changing to a higher-quality cable.

kat-mcg-lin, 2019-05-17:
Just went back and found the original cable it came with, it has the blue inside on the USB, so I'm sure its a USB3. But now when I plug it in, the camera isn't recognized at all. I receive the "No device detected. Is it plugged in?" message.

MartyG-RealSense, 2019-05-17:
The 'No device detected' message sometimes occurs when using the D435i with an Nvidia Jetson TX2 board.  Are you using a TX2, please?

kat-mcg-lin, 2019-05-17:
Nope, its the Jetson Nano Developer Kit

MartyG-RealSense, 2019-05-17:
Ah yes, I re-read your first message and saw that.  My apologies.

I located another case where a user was also having problems with their D435i being detected as USB 2.0 on Nano.

https://github.com/IntelRealSense/librealsense/issues/3759#issuecomment-483862292

kat-mcg-lin, 2019-05-17:
thank you!

I've looked through that and it seems as though it could be an issue with the patches I installed, I will try again.

kat-mcg-lin, 2019-05-21:
This did not work- I installed the patches from an open source (jetson hacks) that was suppose to make the jetson nano compatible with the D435i camera, and yet no change. Does intel have any patches for this camera that I can use?

kat-mcg-lin, 2019-05-22:
@RealSenseCustomerSupport can I please have some assistance on this matter

I have tried so many different ways to get the D435i recognized on the jetson nano, I followed the jetson hacks installation and kernal patches, yet still nothing. I have also updated the FW on the D435i to see if that would help but to no avail

kat-mcg-lin, 2019-05-22:
@MartyG-RealSense @christopherbate @dorodnic

ev-mp, 2019-05-22:
@kat-mcg-lin , what is the FW version installed ? (`rs-enumerate-devices -s`)
Note that D435i had only recently received USB2 support with firmware 5.11.6.200+.
With the older versions it is expected to be recognized as "Intel Realsense USB2" device with no RGB or IMU sensors available.

kat-mcg-lin, 2019-05-22:
@ev-mp that command doesn't work because the device isn't recognized on the jetson at all

kat-mcg-lin, 2019-05-22:
It is version 5.11.4.0 and I just upgraded it yesterday

kat-mcg-lin, 2019-05-22:
the only available firmware on the intel website is 5.11.4- how does one get the 5.11.6?

ev-mp, 2019-05-22:
@kat-mcg-lin hello, USB2 enumerations issues are often produced by hw setup - and trying different cables/USB ports may help. Adding USB3 self-powered hub is also worth mentioning, especially on Low-power systems such as Odroid.
BTW, can you share the platform/hardware you're using as well as the SDK's version?

Regarding fw5.11.6  - it is temporally unavailable (#4038), but resolving the USB2 issue would allow you to utilize the device properly without additional fw upgrades.

kat-mcg-lin, 2019-05-22:
@ev-mp hi!
how do I gather the information about my platform/hardware and SDK version?
I've also tried changing ports and cords, older cords would allow it to be recognized but only as USB 2.1

ev-mp, 2019-05-22:
@kat-mcg-lin, 
- The [Jetson Nano kit](https://developer.nvidia.com/embedded/buy/jetson-nano-devkit) comes with  4 USB3 ports, so it is strange - have you tried to connect any non-Realsense USB3 camera to these ports, and if so - were they recognized as USB3 ? 
Can you share the kernel log when the camera is plugged in ? (`dmesg -wT`)

I would also check the vendor's support site and verify USB BIOS settings.
And in case the configuration is correct then self-powered hub is still a viable option.

At the same time, I'd cross-check that D435i on another preferably desktop PC to rule out unit-related issues.

kat-mcg-lin, 2019-05-23:
I currently don't have a non-Realsense camera, but I have the T265 Realsense tracking camera that is also USB 3 and it gets properly recognized as USB 3.1 

the output for dmesg -wT is attached 
[dmesg -wT output.docx](https://github.com/IntelRealSense/librealsense/files/3212151/dmesg.-wT.output.docx)

As well, the D435i camera is recognized on the host computer I am using, which is a Linux system.

ryan4729, 2019-05-23:
I have D435i working reliably on the Jetson Nano. I'm using firmware 5.11.6.200 with librealsense 2.21.0, and kernel patches are required: https://github.com/jetsonhacksnano/installLibrealsense.

Most cables I've tested don't work, including the one in the box. This one is reliable: https://www.amazon.com/gp/product/B07D7RZ3WV

Hotplugging does not work.

ev-mp, 2019-05-27:
Hello @kat-mcg-lin , according to the log the camera is properly recognized as USB3 device:
>[Wed May 22 13:07:05 2019] usb 2-1.3: **new SuperSpeed USB** device number 4 using tegra-xusb
[Wed May 22 13:07:05 2019] usb 2-1.3: **New USB device found, idVendor=8086, idProduct=0b3a**
[Wed May 22 13:07:05 2019] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed May 22 13:07:05 2019] usb 2-1.3: Product: Intel(R) RealSense(TM) Depth Camera 435i
[Wed May 22 13:07:05 2019] usb 2-1.3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[Wed May 22 13:07:05 2019] usb 2-1.3: SerialNumber: 852323050574
[Wed May 22 13:07:05 2019] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)

In case you haven't changed the configuration then most certainly the inconsistency in USB 2/3 enumeration is due to the HW setup.
Make sure you connect D435i firmly and fast  - slow plug-in may also result in device being enumerated as USB2.

closed 2019-05-28 by kat-mcg-lin