realsenseai/librealsense #4002: How to  align a small region or a point  in depth frame to color frame  using D435 ?

jiang8088, 2019-05-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                             | 
| Platform                                 |  upboard  |
| SDK Version                           |  { librealsense/ 2.<?>.<?> }                       | 
| Language                               |  {C/C++/opencv/pcl/ }                              | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I have a question about depth  and RGB alignment.   Librealsense's  "align sample" can works properly。But ， it took too much time to run the code：“auto processed = align.process(frameset)”. In my project, I just need to align a small area or a point from depth data  to color data，not the whole frame. How can I align a small region or a point from depth   to color . thank you very much!

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi jiang8088,

I am not aware of such way yet, but you can look into the align process and come up with your method(s) for the needs (specifying a smaller area or just a point):
https://github.com/IntelRealSense/librealsense/blob/master/src/proc/align.cpp#L233Hi

Hope this can be helpful.

Thanks!

dorodnic, 2019-06-03:
It makes sense to add this functionality in future releases. We were discussing `crop` processing block that would accomplish similar purpose

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

We plan to implement this, but there is no fix schedule yet. Thanks for your understanding.

RI-DH, 2021-12-17:
48/5000 
I also have the same need to improve the running speed of the program, but I have no idea about how to set ROI. I want to know if you have solved this problem.

RealSenseCustomerSupport, 2021-12-17:
----------------------------------------------

@RI-DH this feature request has not been implemented yet; it is on the internal request list however is prioritized as low. I'm not sure if @jiang8088 or others in the community have developed a work-around.

MartyG-RealSense, 2022-04-23:
Hi @jiang8088  I would like to update you that there is no further progress to report on the feature request at the time of writing this since the last update above in December 2021.

MartyG-RealSense, 2026-10-25:
Case closed due to the significant amount of time that has passed.

closed 2026-10-25 by MartyG-RealSense