realsenseai/librealsense #4004: How to install D435i driver on TX2?

228282416, 2019-05-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | | 
| Operating System & Version |  ubuntu18.04 | 
| Kernel Version (Linux Only)    |                   4.9.140-tegra                      | 
| Platform                                 | NVIDIA Jetson TX2(JetPack4.2)   |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |                | 
| Segment			|                         | 

### Issue Description
I can't install D435i driver on the TX2(JetPack4.2),I have searched some information to solve this problem,but I can't.Who can help me to solve it?Please~T^T

MartyG-RealSense, 2019-05-17:
Advice on setting up D435i on Jetson TX2 is provided on the link below.

https://github.com/IntelRealSense/librealsense/issues/3165#issuecomment-459241459

Tengyun-Mo, 2019-05-23:
I have installed D435i driver on TX2 with flashed Jetpack4.2,may be you need to refer to this link below:
[link](https://github.com/Tengyun-Mo/buildLibrealsense2TX2)

closed 2019-09-06 by RealSenseCustomerSupport