realsenseai/librealsense #4005: crash after pipe.start on Android device

cottes, 2019-05-17:
Hi,

I am using camera D435i with latest FW 5.11.4
I am using librealsense-2.21.0

I have tried to use the example provided on last release of SDK on android studio (C:\Intel\IntelRealSense\librealsense-2.21.0\wrappers\android\examples\native_example)

I have modified it in order to display the depth into an android phone with success. However when I tried to the same thing with the color I have a crash of the application. 

**What I observed is that the pipe.start is blocked:**

_05-17 07:58:34.517   956   956 E DisplayDevice: map unknown ((deprecated) sRGB sRGB Full range)/(RenderIntent::COLORIMETRIC) to default color mode
05-17 07:58:34.517 18873 18914 W librs   :
05-17 07:58:34.521 18873 18914 I librs   : Found UVC Device vid: id-
05-17 07:58:34.521 18873 18914 I librs   : vid- 8086
05-17 07:58:34.521 18873 18914 I librs   : pid- ad6
05-17 07:58:34.521 18873 18914 I librs   : mi- 0
05-17 07:58:34.521 18873 18914 I librs   : unique_id- I
05-17 07:58:34.521 18873 18914 I librs   : path- /dev/bus/usb/003/002
05-17 07:58:34.521 18873 18914 I librs   : susb specification- 210
05-17 07:58:34.521 18873 18914 I librs   : Found 19 unreachable streams, 19 extrinsics deleted
05-17 07:58:34.531 18873 18914 W librs   :
05-17 07:58:34.533 18873 18914 I librs   : Found UVC Device vid: id-
05-17 07:58:34.533 18873 18914 I librs   : vid- 8086
05-17 07:58:34.533 18873 18914 I librs   : pid- ad6
05-17 07:58:34.533 18873 18914 I librs   : mi- 0
05-17 07:58:34.533 18873 18914 I librs   : unique_id- I
05-17 07:58:34.533 18873 18914 I librs   : path- /dev/bus/usb/003/002
05-17 07:58:34.533 18873 18914 I librs   : susb specification- 210
05-17 07:58:34.533   956   956 E DisplayDevice: map unknown ((deprecated) sRGB sRGB Full range)/(RenderIntent::COLORIMETRIC) to default color mode
05-17 07:58:34.533 18873 18914 I librs   : Found 19 unreachable streams, 19 extrinsics deleted
05-17 07:58:34.537  1524  1747 D OpenGLRenderer: endAllActiveAnimators on 0x7001b58900 (RippleDrawable) with handle 0x700145bd20
05-17 07:58:34.538 18873 18914 E librs   : SetXU result:-1_

**my code is quite simple on native-lib.cpp:**

        static const int width = 640, height = 480, fps = 6;

        try {
            rs2::pipeline pipe;
            rs2::config config;
            rs2::colorizer colorizer;

            config.enable_stream(RS2_STREAM_DEPTH, width, height, RS2_FORMAT_Z16, fps);
            config.enable_stream(RS2_STREAM_COLOR, width, height, RS2_FORMAT_RGB8, fps);
            pipe.start(config);
            ...
I have seen one issue where it is mentionned that we should use the v4l2 instead of libuvc but on my side I am directly using the lib librealsense2.so available on https://dl.bintray.com/intel-realsense/librealsense as the example directly downloads it.

for information, the camera is working perfectly on windows. I can capture the depth and the color.

Br,
Stephane

cottes, 2019-05-20:
Hi,
I have recompiled the lib realsense with the following command:
cmake -DANDROID_ABI=arm64-v8a -DCMAKE_TOOLCHAIN_FILE=opt\ndk-r16b\android.toolchain.cmake -DBUILD_SHARED_LIBS=true -DFORCE_LIBUVC=off
and 
cmake -DANDROID_ABI=arm64-v8a -DCMAKE_TOOLCHAIN_FILE=opt\ndk-r16b\android.toolchain.cmake -DBUILD_SHARED_LIBS=true -DFORCE_LIBUVC=on

However it does not fix the issue.
I have still a crash when I am executing the pipe.start when color is activated:

**LIBUVC = OFF**

09-21 14:36:10.800  3704  3748 I librs   : Found UVC Device vid: id-
09-21 14:36:10.800  3704  3748 I librs   : vid- 8086
09-21 14:36:10.800  3704  3748 I librs   : pid- ad6
09-21 14:36:10.800  3704  3748 I librs   : mi- 0
09-21 14:36:10.800  3704  3748 I librs   : unique_id- +
09-21 14:36:10.800  3704  3748 I librs   : path- /dev/bus/usb/003/002
09-21 14:36:10.800  3704  3748 I librs   : susb specification- 210
09-21 14:36:10.805   959   959 E DisplayDevice: map unknown ((deprecated) sRGB sRGB Full range)/(RenderIntent::COLORIMETRIC) to default color mode
09-21 14:36:10.811  3704  3748 W librs   :
09-21 14:36:10.813  3704  3748 F libc    : Fatal signal 6 (SIGABRT), code -6 (SI_TKILL) in tid 3748 (Thread-4), pid 3704 (_native_example)

**LIBUVC = ON**

09-21 15:01:56.204  4215  4284 I librs   : Found UVC Device vid: id-
09-21 15:01:56.204  4215  4284 I librs   : vid- 8086
09-21 15:01:56.204  4215  4284 I librs   : pid- ad6
09-21 15:01:56.204  4215  4284 I librs   : mi- 0
09-21 15:01:56.204  4215  4284 I librs   : unique_id- +
09-21 15:01:56.204  4215  4284 I librs   : path- /dev/bus/usb/003/002
09-21 15:01:56.204  4215  4284 I librs   : susb specification- 210
09-21 15:01:56.215  4215  4284 W librs   :
09-21 15:01:56.216  4215  4284 F libc    : Fatal signal 6 (SIGABRT), code -6 (SI_TKILL) in tid 4284 (Thread-4), pid 4215 (_native_example)

For information, I am using a development board qualcomm based on Snapdragon 845 chip.

Br,
Stephane

matkatz, 2019-05-21:
Hi @cottes,

When you work with the android solution for **unrooted** devices, librealsense doesn't use v4l or libuvc, you can/should let the android studio project to do the required configuration for you.

It seems that your camera is connected via USB 2 port, D435i is currently requires USB 3 connection.
Also the D435i wasn't supported on v2.21.0.

The bottom line, switch to librealsense [v2.22.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.22.0) and connect the camera via USB 3.

BTW IMU streaming is currently not supported on Android, we are working to enable it.

cottes, 2019-05-21:
Hi,
Here is my setup:
- the board is root.
- The USB on the board is USB 3 type C.
- The app is working perfectly when only the depth is selected.

For information I have updated to V2.22.0 but I have the same result:

09-21 15:13:06.200  4708  4752 I  ############################### REAL SENSE ###############################: before pipe.start
09-21 15:13:06.200  4708  4752 I librs   : Found UVC Device vid: id-
09-21 15:13:06.200  4708  4752 I librs   : vid- 8086
09-21 15:13:06.200  4708  4752 I librs   : pid- ad6
09-21 15:13:06.200  4708  4752 I librs   : mi- 0
09-21 15:13:06.200  4708  4752 I librs   : unique_id- /dev/bus/usb/003/002
09-21 15:13:06.200  4708  4752 I librs   : path- /dev/bus/usb/003/002
09-21 15:13:06.200  4708  4752 I librs   : susb specification- 210
09-21 15:13:06.203  4708  4752 E librs   : SetXU result:-1
09-21 15:13:06.207   973   973 E DisplayDevice: map unknown ((deprecated) sRGB sRGB Full range)/(RenderIntent::COLORIMETRIC) to default color mode
09-21 15:13:06.224   973   973 E DisplayDevice: map unknown ((deprecated) sRGB sRGB Full range)/(RenderIntent::COLORIMETRIC) to default color mode
09-21 15:13:06.226  1544  1784 D OpenGLRenderer: endAllActiveAnimators on 0x7d8fc39000 (RippleDrawable) with handle 0x7d8beb50e0
09-21 15:13:06.240   973   973 E DisplayDevice: map unknown ((deprecated) sRGB sRGB Full range)/(RenderIntent::COLORIMETRIC) to default color mode
09-21 15:13:06.256   973   973 E DisplayDevice: map unknown ((deprecated) sRGB sRGB Full range)/(RenderIntent::COLORIMETRIC) to default color mode
09-21 15:13:06.257  4708  4752 W librs   :
09-21 15:13:06.258  4708  4752 F libc    : Fatal signal 6 (SIGABRT), code -6 (SI_TKILL) in tid 4752 (Thread-4), pid 4708 (_native_example)
09-21 15:13:06.274   973   973 E DisplayDevice: map unknown ((deprecated) sRGB sRGB Full range)/(RenderIntent::COLORIMETRIC) to default color mode
09-21 15:13:06.290   973   973 E DisplayDevice: map unknown ((deprecated) sRGB sRGB Full range)/(RenderIntent::COLORIMETRIC) to default color mode

I have a question for you on the USB 2. How do you see on the log that the camera is plugged in USB 2 (SUSB specification). If it is really the case, the camera can select by its own ? If it is really USB2, it could come from the cable. I can check to change it tomorrow.

I would like to add as well that the option -DFORCE_LIBUVC has no effect on the compilation of the librealsense2.so

cottes, 2019-05-23:
I have changed the cable and now it is working. In fact the cable that I have used was USB type C to type C but USB 2. With USB 3, it is better ;-)
Thanks for your support.

matkatz, 2019-05-23:
This is the indication for USB 2:
`09-21 15:13:06.200 4708 4752 I librs : susb specification- 210`

This is what I expect to see when the device is connected via USB3 (on my case the camera is D415 so the pid should be different):
```
2013-01-24 17:48:47.768 4723-4723/com.intel.realsense.camera I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- /dev/bus/usb/004/008
    path- /dev/bus/usb/004/008
    susb specification- 320
2013-01-24 17:48:47.768 4723-4723/com.intel.realsense.camera I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- /dev/bus/usb/004/008
    path- /dev/bus/usb/004/008
    susb specification- 320
```
Forcing libuvc shouldn't have any effect.
Also, unless you manually changed it, the native example downloads librealsense on compile time.

let's check with a different cable.

cottes, 2019-05-23:
I have seen that the example downloaded it automatically but I have modified it to discard the download ;-)

closed 2019-05-24 by cottes