realsenseai/librealsense #4006: Error during Post Processing Filtering in C-API

mmubeen-6, 2019-05-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Linux Mint 19  | 
| Kernel Version (Linux Only)    | 4.15.0-48-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.21.0                          | 
| Language                            |  C                           | 

### Issue Description
Hi, I am currently trying to use the advanced mode on D435 using C API. Currently, I have a working example in which I am able to extract the depth frame along with the RGB Image from the camera without any problem. The code for that is given below:

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

/* Include the librealsense C header files */
#include <librealsense2/rs.h>
#include <librealsense2/rs_advanced_mode.h>
#include <librealsense2/h/rs_advanced_mode_command.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_option.h>
#include <librealsense2/h/rs_frame.h>
#include <librealsense2/h/rs_types.h>
#include <librealsense2/h/rs_processing.h>
#include "../example.h"

#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>

#include <opencv2/highgui/highgui_c.h>
#include <opencv2/core/core_c.h>
#include <opencv2/core/version.hpp>
#include <opencv2/imgproc/imgproc_c.h>
#include <opencv2/videoio/videoio_c.h>

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//                                     These parameters are reconfigurable                                        //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define STREAM_COLOR    RS2_STREAM_COLOR  // rs2_stream is a types of data provided by RealSense device           //
#define FORMAT_COLOR    RS2_FORMAT_RGB8   // rs2_format is identifies how binary data is encoded within a frame   //
#define STREAM_DEPTH    RS2_STREAM_DEPTH  // rs2_stream is a types of data provided by RealSense device //
#define FORMAT_DEPTH    RS2_FORMAT_Z16    // rs2_format is identifies how binary data is encoded within a frame //
#define WIDTH           640               // Defines the number of columns for each frame                         //
#define HEIGHT          480               // Defines the number of lines for each frame                           //
#define FPS             30                // Defines the rate of frames per second                                //
#define STREAM_INDEX    -1                // Defines the stream index, used for multiple streams of the same type //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

int main()
{
    rs2_error* e = 0;

    // Create a context object. This object owns the handles to all connected realsense devices.
    // The returned object should be released with rs2_delete_context(...)
    rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
    check_error(e);

    /* Get a list of all the connected devices. */
    // The returned object should be released with rs2_delete_device_list(...)
    rs2_device_list* device_list = rs2_query_devices(ctx, &e);
    check_error(e);

    int dev_count = rs2_get_device_count(device_list, &e);
    check_error(e);
    printf("There are %d connected RealSense devices.\n", dev_count);
    if (0 == dev_count)
        return EXIT_FAILURE;
[... 155 lines cut ...]
        
        // Converting the extracted frame data to image
        IplImage* ipl_depth_threshold = cvCreateImage(cvSize(width_threshold, height_threshold), IPL_DEPTH_16U, 1);
        ipl_depth_threshold->imageData = (char*) depth_frame_data_threshold;
        
        // Query the distance from the camera to the object in the center of the image
        float dist_to_center = rs2_depth_frame_get_distance(threshold_frame, width_threshold / 2, height_threshold / 2, &e);    check_error(e);
        printf("The camera is facing an object %.3f meters away.\n", dist_to_center);

        cvNamedWindow("Thresholded Depth Frame", CV_WINDOW_AUTOSIZE); 
        cvShowImage("Thresholded Depth Frame", ipl_depth_threshold);

        rs2_release_frame(threshold_frame);
        cvReleaseImage(&ipl_depth_threshold);
}
```
The new code code initially runs perfectly fine for a few frames and shows the appropriate output (atleast thats what I think), but then all of a sudden, it starts to give various error that includes: -

- `Segmentation fault (core dumped)`, this happens around various lines: sometime around the line `rs2_frame* frames = rs2_pipeline_wait_for_frames(pipeline, 5000, &e);` and `rs2_frame* frame_color = rs2_extract_frame(frames, 1, &e);`.

- `rs_error was raised when calling rs2_depth_frame_get_distance(frame_ref:0x562759e56728, x:0, y:0):    Unrecognized depth format 0 bytes per pixel`
This happens sometimes when I pass the `threshold_frame` to the `rs2_depth_frame_get_distance` function. And when this happens, the width and height of the `threshold_frame` turns out both to be 0.

**Note**: My code runs perfectly for some initial 10-15 frames and sometimes even for longer but then eventually crashes every time.
Any help would be appreciated.

ev-mp, 2019-05-20:
@mmubeen-6 hello,
The immedeate observation is that you forget to release the depth frame in the end of the loop
```c++
       ...
        rs2_release_frame(threshold_frame);
       Depth frame release is missing
        cvReleaseImage(&ipl_depth_threshold);
```

The symptoms you reports are also consistent with the default pipeline capacity of 10 frames 
Note the comments in the original section:
```c++
        // **The returned object should be released with rs2_release_frame(...)**
        rs2_frame* frames = rs2_pipeline_wait_for_frames(pipeline, 5000, &e);
        // **The returned object should be released with rs2_release_frame(...)**
        rs2_frame* frame_color = rs2_extract_frame(frames, 1, &e);
        // **The returned object should be released with rs2_release_frame(...)**
        rs2_frame* frame_depth = rs2_extract_frame(frames, 0, &e);
        .....
        rs2_release_frame(frame_depth);
        rs2_release_frame(frame_color);
        rs2_release_frame(frames);
```

mmubeen-6, 2019-05-20:
Hi @ev-mp, I am actually releasing all the frames `rs2_release_frame(frame_depth); rs2_release_frame(frame_color); rs2_release_frame(frames);`. The thing is that the code that does the post-processing filtering has been added along side the previous code (the one that mentioned above it and is working perfectly fine). I could even attach that whole code here if you want me to.

ev-mp, 2019-05-20:
@mmubeen-6 hi, please check it carefully:
The missing part is in the post processing section (see the items in bold):
>   while(1){
        rs2_frame* **frame_depth** = ...                                                         -- This is the first allocation
       ......
        rs2_frame* **threshold_frame** = rs2_wait_for_frame(threshold_queue, 5000, &e);   --- The second allocation
        ...
        ...
        rs2_release_frame(**threshold_frame**);                                        --- The second resource released
      -------------- **_Depth frame handle release is missing_** -------------------
        cvReleaseImage(&ipl_depth_threshold);
}

The other part is provided a reference for proper handling (i.e. the alloc/dealloc are symmetric)

mmubeen-6, 2019-05-22:
Hi @ev-mp, I have done as you told me to do symmetric alloc/dealloc. But I am still getting the same errors after a few frames. 

> - `Segmentation fault (core dumped)`, this happens around various lines: sometime around the line `rs2_frame* frames = rs2_pipeline_wait_for_frames(pipeline, 5000, &e); and rs2_frame* frame_color = [...]

I have attached my complete code, can you please have a look at it. I am compiling it using the command: ``gcc temp.c -lrealsense2 `pkg-config --cflags --libs opencv` -O4 && ./a.out``.

```
/* Include the librealsense C header files */
#include <librealsense2/rs.h>
#include <librealsense2/rs_advanced_mode.h>
#include <librealsense2/h/rs_advanced_mode_command.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_option.h>
#include <librealsense2/h/rs_frame.h>
#include <librealsense2/h/rs_types.h>
#include <librealsense2/h/rs_processing.h>
#include "../example.h"

#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>

#include <opencv2/highgui/highgui_c.h>
#include <opencv2/core/core_c.h>
#include <opencv2/core/version.hpp>
#include <opencv2/imgproc/imgproc_c.h>
#include <opencv2/videoio/videoio_c.h>

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//                                     These parameters are reconfigurable                                        //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define STREAM_COLOR    RS2_STREAM_COLOR  // rs2_stream is a types of data provided by RealSense device           //
#define FORMAT_COLOR    RS2_FORMAT_RGB8   // rs2_format is identifies how binary data is encoded within a frame   //
#define STREAM_DEPTH    RS2_STREAM_DEPTH  // rs2_stream is a types of data provided by RealSense device //
#define FORMAT_DEPTH    RS2_FORMAT_Z16    // rs2_format is identifies how binary data is encoded within a frame //
#define WIDTH           640               // Defines the number of columns for each frame                         //
#define HEIGHT          480               // Defines the number of lines for each frame                           //
#define FPS             30                // Defines the rate of frames per second                                //
#define STREAM_INDEX    -1                // Defines the stream index, used for multiple streams of the same type //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

int main()
{
    rs2_error* e = 0;

    // Create a context object. This object owns the handles to all connected realsense devices.
    // The returned object should be released with rs2_delete_context(...)
    rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
    check_error(e);

    /* Get a list of all the connected devices. */
    // The returned object should be released with rs2_delete_device_list(...)
    rs2_device_list* device_list = rs2_query_devices(ctx, &e);
    check_error(e);

    int dev_count = rs2_get_device_count(device_list, &e);
    check_error(e);
    printf("There are %d connected RealSense devices.\n", dev_count);
    if (0 == dev_count)
        return EXIT_FAILURE;

    // Get the first connected device
    // The returned object should be released with rs2_delete_device(...)
    rs2_device* dev = rs2_create_device(device_list, 0, &e);
    check_error(e);
    print_device_info(dev);

    // Create a pipeline to configure, start and stop camera streaming
    // The returned object should be released with rs2_delete_pipeline(...)
    rs2_pipeline* pipeline =  rs2_create_pipeline(ctx, &e);
[... 119 lines cut ...]
        rs2_release_frame(frames);
        cvReleaseImage(&ipl_frame);
        cvReleaseImage(&ipl_depth);
        cvReleaseImage(&ipl_depth_threshold);
        if (k%256 == 27)
            break;
    }

    // Stop the pipeline streaming
    rs2_pipeline_stop(pipeline, &e);
    check_error(e);

    // Release resources
    rs2_delete_processing_block(threshold_filter);
    rs2_delete_frame_queue(threshold_queue);
    rs2_delete_pipeline_profile(pipeline_profile);
    rs2_delete_config(config);
    rs2_delete_pipeline(pipeline);
    rs2_delete_device(dev);
    rs2_delete_device_list(device_list);
    rs2_delete_context(ctx);

    return EXIT_SUCCESS;
}
```

ev-mp, 2019-05-22:
@mmubeen-6 hello,
There are at least two more places where the code potentially leaks memory:
```c
        // The retunred object should be released with rs2_release_frame(...)
        rs2_frame* frame_depth = rs2_extract_frame(frames, 0, &e);                      check_error(e);
        
        // Check if the given frame can be extended to depth frame interface
        // Accept only depth frames and skip other frames
        if (0 == rs2_is_frame_extendable_to(frame_depth, RS2_EXTENSION_DEPTH_FRAME, &e))
            continue;
        ...
        // Accept only depth frames and skip other frames
        if (0 == rs2_is_frame_extendable_to(threshold_frame, RS2_EXTENSION_DEPTH_FRAME, &e))
           continue;
```
In both cases the flow is interrupted without effectively releasing the allocated resources
In addition, I find such places suspicious :
```c
rs2_frame* frame_color = rs2_extract_frame(frames, 1, &e);
        check_error(e);  
```

There is an API available that allows to query the numbers of frames in the `frameset to avoid dereferencing invalid indexes: `int rs2_embedded_frames_count`
Please check C examples, e.g [rs-distance](https://github.com/IntelRealSense/librealsense/blob/master/examples/C/distance/rs-distance.c#L87-L115)

closed 2020-05-04 by ev-mp