realsenseai/librealsense #4007: T265 device timestamps

VladyslavUsenko, 2019-05-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5502 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |   4.15.0-47-generic   | 
| Platform                                 | PC |
| SDK Version                            |  2.21.0      | 
| Language                            |  c++       | 
| Segment			|  other                | 

### Issue Description
Is it possible to get the device timestamps from T265 with C++? I'm trying to get it with `vf.get_frame_metadata(RS2_FRAME_METADATA_SENSOR_TIMESTAMP)` but get the following error:
```
WARNING [139732104001280] (types.cpp:59) Sensor Timestamp attribute is not applicable for Fisheye stream 
```
The same happens for `RS2_FRAME_METADATA_FRAME_TIMESTAMP`
``` 
WARNING [140379865216768] (types.cpp:59) Frame Timestamp attribute is not applicable for Fisheye stream 
```

MartyG-RealSense, 2019-05-17:
I hope the script in the link below will be useful to you for getting T265 timestamps.

https://github.com/IntelRealSense/librealsense/issues/3738#issuecomment-490033753

VladyslavUsenko, 2019-05-17:
I can get the **system** timestamps. As far as I understand, these are the timestamps at which the data arrived to the PC, right?

What I want to get is a device timestamp (timestamp at which the camera recorded the frame in the camera clock). It looks like the timestamps where changed to system timestamps in #3453, in particular in this commit: 95c1fb8.

The problem is that after this change it is not possible to get the timestamps from `RS2_TIMESTAMP_DOMAIN_HARDWARE_CLOCK`, because they are not provided by the metadata variables `RS2_FRAME_METADATA_SENSOR_TIMESTAMP` and `RS2_FRAME_METADATA_FRAME_TIMESTAMP`.

akirayou, 2019-05-20:
Here is one workaround.
https://github.com/akirayou/librealsense/commit/ecbcb32358dda9535ccd9e72146ea7178311b1ed

But I don't know which is best way
1.Add get_frame_metadata(RS2_FRAME_METADATA_SENSOR_TIMESTAMP)    [I't my workaround]
2.frame time stamp must be HARDWARE_CLOCK [Older T265 (tm-device.cpp) dose this]

Here is related information of realsense-ros.
https://github.com/IntelRealSense/realsense-ros/pull/710

RealSenseCustomerSupport, 2019-08-13:
----------------------------------------------

Hi @VladyslavUsenko,

The PRs below are integrated in 2.25.0. Please check if this fulfill your requirement.
https://github.com/IntelRealSense/librealsense/pull/4448
https://github.com/IntelRealSense/librealsense/pull/4403

closed 2019-08-13 by VladyslavUsenko