realsenseai/librealsense #4008: Support Python3.7

ClimbsRocks, 2019-05-17:
Hi,

As noted in https://github.com/IntelRealSense/librealsense/issues/3110 the PyPi page does not support Python3.7. Is there a specific blocker to supporting Python3.7? 

It was released nearly a year ago at this point, and is the current default recommended version on Python's homepage.

If there is no blocker, when can we expect Python3.7 support?

Thanks,
preston

dorodnic, 2019-05-18:
No blocker, need to update our CI infrastructures. Thanks for raising it.

ClimbsRocks, 2019-05-18:
Awesome, thanks!

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@ClimbsRocks PR#4012"Add python 3.7 support" now merged to v2.23.0. Please have a try and feedback. Thanks!

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@ClimbsRocks Did you get chance to try v2.23? Does it work for you? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-06-21:
----------------------------------------------

@ClimbsRocks Any update with v2.23? Thanks!

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@ClimbsRocks If no other questions about this ticket, will close it accordingly. Thanks!

closed 2019-06-25 by RealSenseCustomerSupport

siddharthcb, 2020-09-16:
hi @MartyG-RealSense 
I tried installing pyrealsense2 for python3.7 on ubuntu18.04 using the following command. 

`python3.7` -m pip install pyrealsense2
Collecting pyrealsense2
  Could not find a version that satisfies the requirement pyrealsense2 (from versions: )
No matching distribution found for `pyrealsense2`

Could you please help me with this?

MartyG-RealSense, 2020-09-16:
The above question by @siddharthcb was addressed in https://github.com/IntelRealSense/librealsense/issues/7352