realsenseai/librealsense #4009: Include could not find load file: libusb_config

ClimbsRocks, 2019-05-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i| 
| Firmware Version                   | Trying to install | 
| Operating System & Version |   Ubuntu 18.04.2 LTS  | 
| Kernel Version (Linux Only)    |  4.15.0-43-generic                                        | 
| Platform                                 | Workstation  |
| SDK Version                            |  Trying to install | 
| Language                            |  python | 
| Segment			|  Robot | 

### Issue Description
As noted https://github.com/IntelRealSense/librealsense/issues/4008 the library does not currently have easy installation available for Python3.7 on PyPi. 

I tried installing from source, and encountered the following error:

```
$ cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true
-- pybind11 v2.2.1
CMake Error at python/CMakeLists.txt:16 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at python/CMakeLists.txt:52 (include):
  include could not find load file:

    libusb_config

-- Configuring incomplete, errors occurred!
```

I tried removing this line, and encountered a different error:
```
$ cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true
-- pybind11 v2.2.1
CMake Error at CMakeLists.txt:16 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at CMakeLists.txt:52 (include):
  include could not find load file:

    libusb_config

-- Configuring incomplete, errors occurred!
```

I tried installing `libusb` but their installation page says it should already be included on most Linux distributions, and package managers did not find it when I tried naively installing `sudo apt install libusb` or `sudo apt install libusb_config`. 

Just to be sure, I installed librealsense both through package manager and from source, and both seemed to install correctly, and were verified using the instructions included in the global install instructions. Under neither installation could I get the errors above to resolve. 

It looks like the last time the relevant code was modified was recently, back in April with https://github.com/IntelRealSense/librealsense/pull/3647 . 

**Can anyone explain how to make `libusb_config` available when building the Python wrapper from source? Or, alternatively, make a Python3.7 version available on PyPi :)**

reynoldscar, 2019-05-17:
I was struggling with the same thing today as well. You need to run the cmake from the highest level directory, librealsense\. The other errors should go away. However, I was working in Windows 10 and using cmake-gui. I additionally downloaded libusb and set LIBUSB_LIB and LIBUSB_INC to the corresponding include and library directories. I also set PYTHON_EXECUTABLE to my desirable python version since I have multiple installed. I am currently building so I am not 100% sure it'll work but it should definitely get you by that and I haven't seen any errors yet. I will update when it finished one way or the other. 

I would also advocate for making a Python3.7 version available on PyPi.

ClimbsRocks, 2019-05-17:
Thanks for the info! Ubuntu comes with `libusb-dev` installed (`libusb-dev is already the newest version (2:0.1.12-31).`), which makes this extra confusing. I couldn't explicitly find any available libraries called `libusb_lib` or `libusb_inc` or `libusb_config`, but if anyone knows a different name to try, I'm happy to.

If you do get this working, I'd love to hear how! I've got a dev python3.6 environment with pyrealsense2 installed from PyPi, and it works great so far. I'm hoping to figure out how to get this installed on py37 though, as that's the environment I've already created with a bunch of finicky deep learning libraries. 

Best of luck in your install- I'm definitely interested to hear how it goes!

reynoldscar, 2019-05-17:
`libusb_lib` and  `libusb_inc` are cmake variables that you need to set. You can most likely set this explicitly in the highest level CMakeLists.txt.  `libusb_config` is a .cmake file.  I was able to get it to successfully build and have python recognize it as a module but now python complains that the module has no attributes, for example, has no attribute 'pipeline' or 'config'. That's where I am stuck now.

ClimbsRocks, 2019-05-18:
Thanks for the update! Hopefully we'll hear from the developers soon. 

@radfordi it looks like you were recently working on a useful PR that touched on libusb. Any thoughts on how we might fix the error:

```
$ cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true
-- pybind11 v2.2.1
CMake Error at python/CMakeLists.txt:16 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at python/CMakeLists.txt:52 (include):
  include could not find load file:

    libusb_config

-- Configuring incomplete, errors occurred!
```

radfordi, 2019-05-20:
Hi @ClimbsRocks,

This is line 16 of `wrappers/python/CMakeLists.txt`.
>  set_target_properties(pyrealsense2 PROPERTIES VERSION
    ${REALSENSE_VERSION_STRING} SOVERSION "${REALSENSE_VERSION_MAJOR}.${REALSENSE_VERSION_MINOR}")

The only way I see that this would have the wrong number of arguments is if `REALSENSE_VERSION_STRING` is not set.

The only way that `libusb_config` would not be found is if this line from the root `CMakeLists.txt` wasn't run:

> list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_LIST_DIR}/CMake)

This error
> python/CMakeLists.txt:16

again makes it seems like you are trying to build `librealsense/wrappers` and not `librealsense`.

All three of these errors lead me to suspect your `cmake ..` should be `cmake ../..`.

radfordi, 2019-05-20:
I verified that these indeed are the errors you get when you point `cmake` at `librealsense/wrappers` instead of `librealsense`.  CC @dorodnic.

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@ClimbsRocks @radfordi With Python3.7 support from v2.23(PR#4012:https://github.com/IntelRealSense/librealsense/pull/4012), do you still have this issue? Looking forward to your update. Thanks!

radfordi, 2019-07-02:
@RealSenseCustomerSupport, this was user error (pointing `cmake` to the wrong directory).  This issue can be closed.

closed 2019-07-02 by RealSenseCustomerSupport