realsenseai/librealsense #4011: T265 pipe.start() leads to RuntimeError:No device connected

Microset, 2019-05-18:
Required Info
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5448 | 
| Operating System & Version  |   Ubuntu 16.04  LTS| 
| Kernel Version (Linux Only)   |  4.15.0-47-generic                                      | 
| Platform                                 | PC  |
| SDK Version                          |   2.20.0                           | 
| Language                              |  python                           | 
| Segment			        |  Robot                          | 

Issue Description
When starting example https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_example.py, I  get error:
  File "t265_example.py", line 21, in <module>
    pipe.start(cfg)
RuntimeError: No device connected

Camera is connected via USB 3.
In the same time everything Ok with a D435, D435i on this PC and the python environment. This T265 working fine with realsese-vewer and rs-enumerate-devices and all other tracking rs-... examples  https://github.com/IntelRealSense/librealsense/tree/master/examples on C++. 
What wrong?
Thank you

dorodnic, 2019-05-18:
If python module was built from source, it is common for T265 support to get disabled during CMake due to invalid proxy settings. You can check CMake output to make sure or try the `pip` version if its available on your platform. Another reason may be that the installed python module is from an older version of librealsense. Finally, `pip install` does not update `udev` rules, this is only done by `librealsense2-dkms` package and the `udev` script. You might need to re-apply the udev rules from latest version.

Microset, 2019-05-20:
@dorodnic, thank you!
We did the following:
1) "make uninstall" from our previous librealsense source build directory;
2) installed pre-build packages according https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.
Result is the same:
1) All C++ examples and our python programs for D435 work;
2) T265 C++ examples work;
3) T265 python examples  from https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples finish with "no  device connected" error.

Microset, 2019-05-21:
@dorodnic, thank you!
The problem has been solved after installation librealsense v2.22

closed 2019-05-21 by Microset