realsenseai/librealsense #4014: realsense sr300 error of the sdk

TFLQW, 2019-05-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR300 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { v2.21.0}                          | 
| Language                            |  {C }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
when i run the sdk of the realsense i try to turn on the code-light depth sensor is was not work and pull up this error,but the rgb camera is worked
xioctl(VIDIOC_DQBUF) failed for fd: 22 Last Error: No such device

ev-mp, 2019-05-19:
@TFLQW hello, 
The errors during `xioctl(VIDIOC_DQBUF)` calls are often associated with device being disconnected due to insufficient power or similar setup issue.
Can you share the kernel log when the event occurs ? (`dmesg -wT`) 

In case you encounter 'device disconnected' messages, then you may mitigate this by changing the USB port/cable and/or switch to a self-powered USB3 hub .

TFLQW, 2019-05-21:
@ev-mp 
The log message is below and i try the D415 camera and export no errors, i think i have miss some portion of the SR300.
[二 5月 21 20:46:40 2019] microcode: microcode updated early to revision 0x25, date = 2019-02-26
[二 5月 21 20:46:40 2019] Linux version 4.15.0-45-generic (buildd@lcy01-amd64-027) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.10)) #48~16.04.1-Ubuntu SMP Tue Jan 29 18:03:48 UTC 2019 (Ubuntu 4.15.0-45.48~16.04.1-generic 4.15.18)
[二 5月 21 20:46:40 2019] Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-45-generic root=UUID=d5a3101b-9017-4054-9002-bbed40884b8c ro quiet splash vt.handoff=7
[二 5月 21 20:46:40 2019] KERNEL supported cpus:
[二 5月 21 20:46:40 2019]   Intel GenuineIntel
[二 5月 21 20:46:40 2019]   AMD AuthenticAMD
[二 5月 21 20:46:40 2019]   Centaur CentaurHauls
[二 5月 21 20:46:40 2019] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[二 5月 21 20:46:40 2019] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[二 5月 21 20:46:40 2019] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[二 5月 21 20:46:40 2019] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[二 5月 21 20:46:40 2019] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[二 5月 21 20:46:40 2019] e820: BIOS-provided physical RAM map:
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x0000000000059000-0x000000000009efff] usable
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x0000000000100000-0x00000000bacc3fff] usable
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000bacc4000-0x00000000baccafff] ACPI NVS
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000baccb000-0x00000000bb0f5fff] usable
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000bb0f6000-0x00000000bb6f1fff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000bb6f2000-0x00000000ca85dfff] usable
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000ca85e000-0x00000000cabaafff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000cabab000-0x00000000cabc8fff] ACPI data
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000cabc9000-0x00000000cb56efff] ACPI NVS
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000cb56f000-0x00000000cbf6cfff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000cbf6d000-0x00000000cbffdfff] type 20
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000cbffe000-0x00000000cbffefff] usable
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000cd000000-0x00000000cf1fffff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[二 5月 21 20:46:40 2019] BIOS-e820: [mem 0x0000000100000000-0x000000012fdfffff] usable
[二 5月 21 20:46:40 2019] NX (Execute Disable) protection: active
[二 5月 21 20:46:40 2019] efi: EFI v2.31 by American Megatrends
[二 5月 21 20:46:40 2019] efi:  ACPI=0xcabb1000  ACPI 2.0=0xcabb1000  SMBIOS=0xf04c0  MPS=0xfd830 
[二 5月 21 20:46:40 2019] secureboot: Secure boot could not be determined (mode 0)
[二 5月 21 20:46:40 2019] SMBIOS 2.8 present.
[二 5月 21 20:46:40 2019] DMI: Dell Inc. Inspiron 3542/0926J6, BIOS A14 05/23/2018
[二 5月 21 20:46:40 2019] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[二 5月 21 20:46:40 2019] e820: remove [mem 0x000a0000-0x000fffff] usable
[二 5月 21 20:46:40 2019] e820: last_pfn = 0x12fe00 max_arch_pfn = 0x400000000
[二 5月 21 20:46:40 2019] MTRR default type: uncachable
[二 5月 21 20:46:40 2019] MTRR fixed ranges enabled:
[二 5月 21 20:46:40 2019]   00000-9FFFF write-back
[二 5月 21 20:46:40 2019]   A0000-BFFFF uncachable
[二 5月 21 20:46:40 2019]   C0000-CFFFF write-protect
[二 5月 21 20:46:40 2019]   D0000-DFFFF uncachable
[二 5月 21 20:46:40 2019]   E0000-FFFFF write-protect
[二 5月 21 20:46:40 2019] MTRR variable ranges enabled:
[二 5月 21 20:46:40 2019]   0 base 0000000000 mask 7F00000000 write-back
[二 5月 21 20:46:40 2019]   1 base 0100000000 mask 7FE0000000 write-back
[二 5月 21 20:46:40 2019]   2 base 0120000000 mask 7FF0000000 write-back
[二 5月 21 20:46:40 2019]   3 base 00E0000000 mask 7FE0000000 uncachable
[二 5月 21 20:46:40 2019]   4 base 00D0000000 mask 7FF0000000 uncachable
[二 5月 21 20:46:40 2019]   5 base 00CE000000 mask 7FFE000000 uncachable
[二 5月 21 20:46:40 2019]   6 base 00CD000000 mask 7FFF000000 uncachable
[二 5月 21 20:46:40 2019]   7 base 012FE00000 mask 7FFFE00000 uncachable
[二 5月 21 20:46:40 2019]   8 disabled
[二 5月 21 20:46:40 2019]   9 disabled
[二 5月 21 20:46:40 2019] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[二 5月 21 20:46:40 2019] total RAM covered: 4046M
[二 5月 21 20:46:40 2019] Found optimal setting for mtrr clean up
[二 5月 21 20:46:40 2019]  gran_size: 64K 	chunk_size: 4M 	num_reg: 8  	lose cover RAM: 0G
[二 5月 21 20:46:40 2019] e820: update [mem 0xcd000000-0xffffffff] usable ==> reserved
[... 596 lines cut ...]
[... cut ...]
[二 5月 21 20:46:41 2019] input: Synaptics s3203 as /devices/pci0000:00/INT33C3:00/i2c-1/i2c-DLL0652:00/0018:06CB:2985.0001/input/input5
[二 5月 21 20:46:41 2019] hid-rmi 0018:06CB:2985.0001: input,hidraw0: I2C HID v1.00 Mouse [DLL0652:00 06CB:2985] on i2c-DLL0652:00
[二 5月 21 20:46:41 2019] nouveau 0000:08:00.0: fb: 2048 MiB DDR3
[二 5月 21 20:46:41 2019] usb 1-1: new high-speed USB device number 2 using ehci-pci
[二 5月 21 20:46:41 2019] nouveau 0000:08:00.0: volt: couldn't find speedo value, volting not possible
[二 5月 21 20:46:41 2019] nouveau 0000:08:00.0: bus: MMIO read of 00000000 FAULT at 612004 [ IBUS ]

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

Hi @TFLQW,

Not saw SR300 enumerated in the kernel log. How about the command "lsusb -v"?

RealSenseCustomerSupport, 2019-06-20:
----------------------------------------------

Hi @TFLQW,

Do you still have trouble about this issue? or, try another PC to see how it works? If the issue clarified, will close this.

TFLQW, 2019-06-21:
I still cannot solve this problem!

TFLQW, 2019-06-21:
@RealSenseCustomerSupport I still cannot solve this problem! When i run "lsusb -v" the information as follow
us 002 Device 006: ID 041e:409b Creative Technology, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x041e Creative Technology, Ltd
  idProduct          0x409b 
  bcdDevice           10.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          150
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength           40
        bInCollection           1
        baInterfaceNr( 0)       1
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0205 Microphone Array
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 2
        bSourceID               1
        bControlSize            1
        bmaControls( 0)      0x01
          Mute Control
        bmaControls( 1)      0x02
          Volume Control
[... 79 lines cut ...]
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     124
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0025  1x 37 bytes
        bInterval               1

RealSenseCustomerSupport, 2019-06-21:
----------------------------------------------

Hi @TFLQW,

It looks the info is missing a lot from "lsusb -v". I can't see SR300 USB device listed.
1. Please only try to list the device only using "lsusb" and run "rs-enumerate-devices" app.
2. Try below command and show the result.
$ sudo apt install v4l-utils
$ v4l2-ctl --list-devices

TFLQW, 2019-06-21:
@RealSenseCustomerSupport After the first step i get as follow
tiboy@tiboy-pc:~$ lsusb
Bus 001 Device 006: ID 0cf3:0036 Atheros Communications, Inc. 
Bus 001 Device 004: ID 1bcf:2b8b Sunplus Innovation Technology Inc. 
Bus 001 Device 003: ID 045e:07b2 Microsoft Corp. 
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 8086:0aa5 Intel Corp. 
Bus 003 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 041e:409b Creative Technology, Ltd 
Bus 002 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
tiboy@tiboy-pc:~$ rs-enumerate-devices 
Device info: 
    Name                          : 	Intel RealSense SR300
    Serial Number                 : 	612201001455
    Firmware Version              : 	3.21.0.0
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.1/3-1.1:1.2/video4linux/video2
    Debug Op Code                 : 	53
    Product Id                    : 	0AA5

Stream Profiles supported by Coded-Light Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  640x480	@ 60Hz	   Y16
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y16
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 10Hz	   Y16
    Infrared 1	  640x480	@ 10Hz	   Y8
    Infrared 1	  640x240	@ 110Hz	   Y16
    Infrared 1	  640x240	@ 110Hz	   Y8
    Infrared 1	  640x240	@ 60Hz	   Y16
    Infrared 1	  640x240	@ 60Hz	   Y8
    Infrared 1	  640x240	@ 30Hz	   Y16
    Infrared 1	  640x240	@ 30Hz	   Y8
    Infrared 1	  640x240	@ 10Hz	   Y16
    Infrared 1	  640x240	@ 10Hz	   Y8
    Depth	  640x480	@ 60Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 10Hz	   Z16
    Depth	  640x240	@ 110Hz	   Z16
    Depth	  640x240	@ 60Hz	   Z16
    Depth	  640x240	@ 30Hz	   Z16
    Depth	  640x240	@ 10Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1920x1080	@ 30Hz	   RGB8
    Color	  1920x1080	@ 30Hz	   Y16
    Color	  1920x1080	@ 30Hz	   BGRA8
    Color	  1920x1080	@ 30Hz	   RGBA8
    Color	  1920x1080	@ 30Hz	   BGR8
    Color	  1920x1080	@ 30Hz	   YUYV
    Color	  1920x1080	@ 10Hz	   RGB8
    Color	  1920x1080	@ 10Hz	   Y16
    Color	  1920x1080	@ 10Hz	   BGRA8
    Color	  1920x1080	@ 10Hz	   RGBA8
    Color	  1920x1080	@ 10Hz	   BGR8
    Color	  1920x1080	@ 10Hz	   YUYV
    Color	  1280x720	@ 60Hz	   RGB8
    Color	  1280x720	@ 60Hz	   Y16
    Color	  1280x720	@ 60Hz	   BGRA8
    Color	  1280x720	@ 60Hz	   RGBA8
    Color	  1280x720	@ 60Hz	   BGR8
    Color	  1280x720	@ 60Hz	   YUYV
    Color	  1280x720	@ 30Hz	   RGB8
    Color	  1280x720	@ 30Hz	   Y16
[... 111 lines cut ...]
    Color	  320x240	@ 30Hz	   YUYV
    Color	  320x240	@ 10Hz	   RGB8
    Color	  320x240	@ 10Hz	   Y16
    Color	  320x240	@ 10Hz	   BGRA8
    Color	  320x240	@ 10Hz	   RGBA8
    Color	  320x240	@ 10Hz	   BGR8
    Color	  320x240	@ 10Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 10Hz	   RGB8
    Color	  320x180	@ 10Hz	   Y16
    Color	  320x180	@ 10Hz	   BGRA8
    Color	  320x180	@ 10Hz	   RGBA8
    Color	  320x180	@ 10Hz	   BGR8
    Color	  320x180	@ 10Hz	   YUYV

TFLQW, 2019-06-21:
@RealSenseCustomerSupport After the second step i get 
tiboy@tiboy-pc:~$ v4l2-ctl --list-devices
Intel(R) RealSense(TM) Camera S (usb-0000:00:14.0-1.1):
	/dev/video1
	/dev/video2

Integrated_Webcam_HD: Integrate (usb-0000:00:1d.0-1.5):
	/dev/video0

closed 2019-07-02 by TFLQW